SpacehuhnTech/WiFiDuck #3: It's giving me an error while compiling.

glblduh, 2019-10-14:
The error is
invalid conversion from 'const char*' to 'uint8_t {aka unsigned char}' [-fpermissive]

This is the log

sketch\cli.cpp: In lambda function:

sketch\cli.cpp:112:29: warning: passing NULL to non-pointer argument 1 of 'String::String(int, unsigned char)' [-Wconversion-null]

             String s { NULL };

                             ^

sketch\spiffs.cpp: In function 'void spiffs::write(String, const char*)':

spiffs.cpp:109:24: error: invalid conversion from 'const char*' to 'uint8_t {aka unsigned char}' [-fpermissive]

             f.write(str);

                        ^

In file included from sketch\spiffs.h:10:0,

                 from sketch\spiffs.cpp:7:

C:\Users\gilla\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266/FS.h:54:12: error:   initializing argument 1 of 'virtual size_t fs::File::write(uint8_t)' [-fpermissive]

     size_t write(uint8_t) override;

            ^

sketch\spiffs.cpp: In function 'void spiffs::streamWrite(const char*, size_t)':

spiffs.cpp:153:50: error: invalid conversion from 'const char*' to 'const uint8_t* {aka const unsigned char*}' [-fpermissive]

         if (streamFile) streamFile.write(buf, len);

                                                  ^

In file included from sketch\spiffs.h:10:0,

                 from sketch\spiffs.cpp:7:

C:\Users\gilla\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.0.0\cores\esp8266/FS.h:55:12: error:   initializing argument 1 of 'virtual size_t fs::File::write(const uint8_t*, size_t)' [-fpermissive]

     size_t write(const uint8_t *buf, size_t size) override;

            ^

exit status 1
invalid conversion from 'const char*' to 'uint8_t {aka unsigned char}' [-fpermissive]

spacehuhn, 2019-10-14:
Hi,
it looks like you're compiling the ESP8266 code with an outdated Arduino Core. 
It's very important to have the latest installed and selected. [Requirements](https://github.com/spacehuhn/WiFiDuck#flash-software)  

My [deauther](https://github.com/spacehuhn/esp8266_deauther) firmware needs the 2.0.0 version that you have installed, because newer versions only allow packet injection of certain types.  
This project uses features and improvements only available in later versions. 
Most importantly the [Async](https://github.com/me-no-dev/ESPAsyncTCP) libraries.  

I guess the README.md doesn't make that detail clear enough, I will update it!

[image]

glblduh, 2019-10-15:
It works now. Thanks. I love all of your projects :D

By the way, does this project connections can be the same from the original wifi ducky? Or it this need a new connection?

spacehuhn, 2019-10-15:
I'm glad it works now :)

The wired connection between the atmega and esp8266 are different if you mean that. 
The old ducky used serial, now it's i2c.

glblduh, 2019-10-15:
Oh okay. Thank you.

closed 2019-10-15 by glblduh