SpacehuhnTech/WiFiDuck #5: [Bug] i2c connection problem

glblduh, 2019-10-16:
It gives an i2c connection problem when I run a script with this line

`STRING msedge https://youtube.com`

When that error came I can't run scripts until I replug the board.

spacehuhn, 2019-10-16:
**If you wish to report a bug, you need to provide more information.
I added the bug issue template for a reason:**

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 1.0]
 - Hardware [e.g. DSTIKE, DIY]

**Additional context**
Add any other context about the problem here.

glblduh, 2019-10-16:
Oh okay. Do I need to open another or just edit the one on the top?

spacehuhn, 2019-10-16:
No, just write everything down here. 
Nobody can help you with "It gives an i2c connection problem".

What hardware are you using?
When does it happen?
Did it work before?
Does it happen everytime?
Do other scripts/commands work?
....

glblduh, 2019-10-16:
Hardware is a NodeMCU v2 and an Arduino Pro Micro. It happens every time. It didn't work before. Other scripts do work.

spacehuhn, 2019-10-16:
have you set a default delay in your script?

glblduh, 2019-10-16:
Yes but it still doesn't work

glblduh, 2019-10-16:
This is  from the serial of the NodeMCU

DEFAULTDELAY 300
> started "/example"
NEW STATUS [5] = PROCESSING
NEW STATUS [255] = PROCESSING
Message from 1 [6 byte]
status
running /exampleMessage from 1 [6 byte]
status
running /exampleNEW STATUS [45] = PROCESSING
NEW STATUS [0] = OK
Sending: GUI r

GUI r
NEW STATUS [5] = PROCESSING
NEW STATUS [255] = PROCESSING
Message from 1 [6 byte]
status
running /exampleNEW STATUS [45] = PROCESSING
Message from 1 [6 byte]
status
running /exampleNEW STATUS [0] = OK
Sending: STRING chrome https://youtube.com

STRING chrome https://youtube.com
Request error :(
NEW STATUS [254] = ERROR
Stopped script
Message from 1 [6 byte]
status
i2c connection problemMessage from 1 [6 byte]
status
i2c connection problem

deantonious, 2019-10-17:
> **If you wish to report a bug, you need to provide more information. I added the bug issue template for a reason:**
>
> **Describe the bug**
> A clear and concise description of what the bug is.
>
> **To [...]

Could you please provide this information?

glblduh, 2019-10-17:
> @glblduh can you try to get the serial output from the pro micro?
> It's annoying but you have to connect it, open the serial monitor and then reflash it and wait until it restarts to be able to see the serial output.

Okay

glblduh, 2019-10-17:
This is what I got

`RECEIVE
Received packet
I2C REQUEST
Interpreting: DEFAULTDELAY 200
RECEIVE
Received packet
I2C REQUEST`

spacehuhn, 2019-10-18:
Can you try to edit [`esp_duck/config.h` line 31](https://github.com/spacehuhn/WiFiDuck/blob/123fc46e8d80440ac304de3bf6c5d66b0121fc2f/esp_duck/config.h#L31) to say `#define I2C_CLOCK_SPEED 50000L` and flash that? Maybe even set the number lower to `10000L`. 
I would just like to know if that has any impact on your setup.

glblduh, 2019-10-19:
So I just broke my Arduino Pro Micro because of my impatience. I accidentally ripped out a capacitor and a resistor. I will be buying another one again next week. I'll be trying your again project next next week.

closed 2019-10-19 by glblduh