SpacehuhnTech/WiFiDuck #8: Internal connection problem [Help]

Jilickpock, 2019-11-04:
[image]

Anytime I attempt to run a script I am given this error and nothing is typed. I've tried multiple boards with no success. Am currently using Leonardo with D1 mini.

spacehuhn, 2019-11-04:
Seems like the I2c connection has an issue.
Can you show us how you connected them?
I can try to recreate it. Got a Leonardo lying around somewhere.

Jilickpock, 2019-11-05:
[image]
[image]
Thanks for the help!
EDIT: Also have tried different wires

spacehuhn, 2019-11-05:
I can't really see if SDA and SCL are connected properly because both wires are orange.
Just for the sakes of it, try switching SCA and SCL and try other jumper cables too. 
I wasted like an hour debugging before and it just turned out one jumper wire was broken.

Also try to connect 5V to 5V instead of 3.3V to 3V3. 
I found that the ESP is better of getting 5V and using it's own voltage regulator instead of sharing the power supply of the other board. That of course is only possible because the Wemos D1 Mini has everything on board and all this can differ from board to board. 
So might not help you but doesn't hurt to try it out, hardware sometimes works in mysterious ways.

Jilickpock, 2019-11-05:
So I tried with all new wires again, and I tried switching it to 5V and still nothing. Any other ideas? Here's another picture of the current layout:

[image]

spacehuhn, 2019-11-05:
Connect the leonardo, open the atmega_duck.ino, open the serial monitor with baud 115200 and flash it again. Then try to run a script again over the web interface and paste the serial output here.

Jilickpock, 2019-11-05:
There isn't any output in the serial monitor. Which I'm guessing means there's still a connection problem. But I have no idea how that'd be possible

spacehuhn, 2019-11-05:
When you have the sketch open, go to `config.h` and uncomment line 17 (https://github.com/spacehuhn/WiFiDuck/blob/master/atmega_duck/config.h#L17)

I'm working on making this whole process easier, sorry you have to go through all of this.

Jilickpock, 2019-11-05:
After making the change and trying to flash it I get an error `'DEBUG_PORT' was not declared in this scope`
Pointing to https://github.com/spacehuhn/WiFiDuck/blob/master/atmega_duck/debug.h#L16

Jilickpock, 2019-11-05:
Wow... ummm don't know what happend but it started working.... I updated to the latest code (2 days ago) I'm gonna try it for a bit and push it. See if the error comes back. Maybe it was a connection issue and a wire was bent just right. Looking back I was using an old one:
[WiFiDuck-master.zip](https://github.com/spacehuhn/WiFiDuck/files/3810823/WiFiDuck-master.zip)

spacehuhn, 2019-11-05:
Oh yes sorry you need to uncomment line 17, 18 **and** 19. 

I also worked on the repo the past days so that might have played a role in it. 
Still working heavily on the software at the moment.

Let me know if you encounter more problems!

closed 2019-11-05 by Jilickpock