SpacehuhnTech/WiFiDuck #9: webinterface dont interact [Help]

ib0ts, 2019-11-07:
I buyed the preflashed wifiduck on [tindie](https://www.tindie.com/products/lspoplove/dstike-wifi-duck/#specs). So i puged it in and connected to the wifi. It seams to be fine the site comes up i can see the main page, terminal, settings. But anytime i click on the run or save button nothing happens. See in the screenshot there is no request send to the ducky. So i tried to add a file on the web-terminal. There i can see the output of the other commands but either not able to crate a new file.
[image]
[image]

spacehuhn, 2019-11-07:
It appears that the ESP8266 firmware wasn't compiled with a SPIFFS partition.
I'm sorry that shouldn't happen on the preflashed devices. 

You will need to reflash the ESP8266. 
I'm working on simplifying that process a lot. I'm afraid though that for now, you will have to follow the README and go through the flashing process. 

But don't worry, I can compile and send you a bin file later that you just have to upload at 192.168.4.1/settings.html

spacehuhn, 2019-11-07:
There you go. Unzip it and upload the bin file at the settings page.

[esp_duck.ino.dstike.bin.zip](https://github.com/spacehuhn/WiFiDuck/files/3819651/esp_duck.ino.dstike.bin.zip)

It will flash itself with that firmware. That process can take a minute or so. 
After it restarted, it will have to format the file system, that can also take a minute or two.
So just wait a bit. If the LED turns green, everything should be good to go.

ib0ts, 2019-11-07:
Okey thank you very much i will test it.

spacehuhn, 2019-11-07:
I forgot that I added the over the air update feature later on.
Is there even a upload button on your settings page 😄 ?

If not you will need to flash it the way it's explained in the README and set the [DIPswitch](https://github.com/spacehuhn/wifiduck#dstike-wifi-duck). 

Or if you're not in a hurry, you can wait a bit until I finished the Arduino board package that will make the process over Arduino a lot easier.

zax400, 2019-11-08:
is posible that included the keyboard spanish language in next versions?, thanks

closed 2019-11-11 by ib0ts