SpacehuhnTech/WiFiDuck #14: [Feature] Change Device type

jLynx, 2019-11-12:
It would be awesome to add the functionality to be able to change the device type/PID live without needing to re program it with arduino studio.

I have a few ideas on how this can be done.
- It may be possible to do it directly in code which would be the easiest way to do it
- Have a way to flash flash the device (Like an OTA which you are currently working on) but being able to select the type you want there (Appley keyboard, Microsoft keyboard, etc...). Then it will just reflash with the ID you want

spacehuhn, 2019-11-22:
So this is possible now due to the custom Arduino board package.
As far as I know, this can't be changed dynamically though, you'll need to recompile the code.

jLynx, 2019-11-22:
But you know how to can flash it on the Web interface, can you flash it with a new device id?

spacehuhn, 2019-11-22:
The Atmega can't be flashed OTA :(

jLynx, 2019-11-22:
😢

closed 2020-02-27 by stale[bot]