SpacehuhnTech/WiFiDuck #17: [Bug] string doesn't finish typing before next command

jLynx, 2019-11-12:
Having an issue where I have a string that is types out followed by another command. In this case `ENTER`.
What happens is it types out ENTER instead of pressing the enter key.
It only types out: https://grabify.link/8ESENTER
Where it should be typing out https://grabify.link/8ESCC1

But splitting them onto 2 lines solves the issue e.g
`DEFAULTDELAY 100
GUI r
STRING https://grabify.link/
STRING 8ESCC1
ENTER`

jLynx, 2019-11-12:
To recreate this issue, it only seems to happen sometimes after power on.

The easyest way to test it is to set the script to auto start and unplug and plug in the devce till you see it happen. should not take more than 10 tries

Jilickpock, 2019-11-13:
I've also encountered this issue. For some reason I've only really seen it when I am using "gui r"
Maybe that has something to do with it?

spacehuhn, 2019-11-16:
So I've been trying to recreate this issue but it just works for me.... every time 😅 
On both my DIY and DSTIKE boards.

Does the issue still persists with the current codebase?

The only cause I can think of is a glitch in the communication between ESP and Atmega.

jLynx, 2019-11-16:
I'll test master again soon. But I'm pretty sure it will still be there. I'll update you on this soon

jLynx, 2019-11-16:
Well I have just updated and I cant reproduce this issue anymore. @Jilickpock  do you still have this issue?

spacehuhn, 2019-11-16:
A week ago or so, I updated the communication module on both ends. That might have fixed it. 
The issue was created afterwards, but maybe you had still the old code running? 
Thats the only explaination I could come up with. I'm really just guessing here.

spacehuhn, 2019-11-17:
Actually I just found some problems in the communication module. 
I'll refactore the entire thing...

Jilickpock, 2019-11-17:
Ya. I just updated. I don't think it happens as much but if I run a script from the scripts section (without hitting the edit button) it still happens.

spacehuhn, 2019-11-19:
This seems to be a timing issue on the I2C bus. 
I made a branch for the fix: https://github.com/spacehuhn/WiFiDuck/tree/com-refactor
But I will have to make some more changes before this can be merged.

spacehuhn, 2019-11-20:
Can you test the branch?
You will need to update both Atmega and ESP.
My quick tests show that everything is working, but at this point I feel very unsure about just merging it right away.

jLynx, 2019-11-21:
Tested it and it seems fine now

spacehuhn, 2019-11-21:
Merged :)

closed 2019-11-21 by spacehuhn