SpacehuhnTech/WiFiDuck #28: [Help]  Internal Connection Problem, can't run certain scripts, unable to update firmware

cameronus, 2019-12-10:
**Error:**
I bought a fresh DSTIKE WiFi Duck, plugged it in, followed the basic instructions to connect to its network, change the SSID and password, and writing a script. Whenever I load the main page, an "Internal Connection Problem" banner appears at the top of the screen. I then attempted to run the following script, but it did not execute:
```
REM Hello World
DEFAULTDELAY 200
GUI SPACE
STRING terminal
ENTER
STRING ls
```
Despite these two issues, the following script *will* execute properly (though the internal connection problem banner remains):
```
STRING test
```

**Environment:**
 - OS: macOS (target/victim), iOS (wifi duck console)
 - Browser: Google Chrome (target/victim), Safari (wifi duck console)
 - Hardware: DSTIKE 1.0 software

**Attempts:**
I tried updating to the latest firmware, but was prevented from flashing to the ATMEGA due to an error:
```
Arduino: 1.8.10 (Mac OS X), Board: "DSTIKE WiFi Duck (ATmega32u4), Arduino Leonardo, Serial (115200b, newline)"

Sketch uses 14384 bytes (50%) of program storage space. Maximum is 28672 bytes.
Global variables use 1765 bytes (68%) of dynamic memory, leaving 795 bytes for local variables. Maximum is 2560 bytes.

avrdude: ser_recv(): read error: Device not configured
avrdude: butterfly_recv(): programmer is not responding
avrdude: ser_drain(): read error: Device not configured
avrdude: ser_send(): write error: Device not configured
avrdude: ser_recv(): read error: Device not configured
avrdude: butterfly_recv(): programmer is not responding
avrdude: ser_send(): write error: Device not configured
avrdude: ser_recv(): read error: Device not configured
avrdude: butterfly_recv(): programmer is not responding
avrdude: ser_recv(): read error: Device not configured
avrdude: butterfly_recv(): programmer is not responding
avrdude: ser_send(): write error: Device not configured
avrdude: ser_recv(): read error: Device not configured
avrdude: butterfly_recv(): programmer is not responding
Found programmer: Id = ""; type =  
    Software Version = . ; Hardware Version =  . 
avrdude: ser_send(): write error: Device not configured
avrdude: ser_recv(): read error: Device not configured
avrdude: butterfly_recv(): programmer is not responding
avrdude: ser_send(): write error: Device not configured
avrdude: ser_recv(): read error: Device not configured
avrdude: butterfly_recv(): programmer is not responding
avrdude: error: buffered memory access not supported. Maybe it isn't
a butterfly/AVR109 but a AVR910 device?
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: ser_send(): write error: Device not configured
avrdude: ser_recv(): read error: Device not configured
avrdude: butterfly_recv(): programmer is not responding
avrdude: error: programmer did not respond to command: leave prog mode
avrdude: ser_send(): write error: Device not configured
avrdude: ser_recv(): read error: Device not configured
avrdude: butterfly_recv(): programmer is not responding
avrdude: error: programmer did not respond to command: exit bootloader
avrdude: ser_close(): can't reset attributes for device: Device not configured
the selected serial port avrdude: ser_close(): can't reset attributes for device: Device not configured
 does not exist or your board is not connected

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
```

spacehuhn, 2019-12-10:
Thanks for the comprehensive description. 

I guess you received one of the earlier boards that aren't yet flashed with the latest version.

Have you watched my video explaining the update process?
https://www.youtube.com/watch?v=e3-nsOjclsY

If the Atmega flashing problems persist, please describe your procedure getting you to the error in more detail. Right now I could only guess that maybe the Atmega is bad, USB connection is bad, drivers are missing (but should be installed with Arduino) or the wrong serial COM port was selected.

spacehuhn, 2019-12-12:
Someone messaged me with the same error on macOS. I have a old Macbook lying around and it works there, but I didn't test on macOS Catalina. So maybe it could be related to that.

spacehuhn, 2019-12-12:
Seems like a big issue that's still not solved: https://github.com/arduino/Arduino/issues/9290

cameronus, 2019-12-12:
I'm carefully following the procedure of downloading the board files and selecting the correct board config, port, etc. While the flashing issue might be due to the OS I'm running, that still does not explain the bugs I'm encountering on the WiFi Duck software, right?

cameronus, 2019-12-12:
I successfully updated both the ATMega and the ESP and everything works perfectly. It seems that it was a Mac driver issue that was fixed by the update. Thanks for your help!

closed 2019-12-12 by cameronus