SpacehuhnTech/WiFiDuck #30: ATMEGA32U4 ESP8266 - Steps to be followed

cnu77, 2020-01-21:
I have the ATMEGA32U4 ESP8266, which has both the Microcontroller part and the WIFI part in a single board. How do I use these scripts in this case?

Do I need to follow the same steps?

spacehuhn, 2020-01-21:
Please provide more information about the board you like to use. The steps are the same, but maybe you need to make some mofications to compile it. That depends on how the two chips are connected to each other (which pins).

cnu77, 2020-01-22:
[image]
[image]
Sorry, I do not have technical knowledge. But these 2 images attached...I have bought that.

alican, 2020-02-04:
Hi @cnu77, Hi @spacehuhn 
I have the same board and I tried the following steps: https://github.com/puckk/CJMCU-3212
More information is here: https://github.com/spacehuhn/wifi_ducky/issues/91
or here: https://www.youtube.com/watch?v=G-Z2rf_AN-c
Is for the old ducky version but with the modifications it worked fine and I flashed the Atmega32u4 and ESP8266.
But the problem is, that I get an "internal connection error" when I run a script.
Probably the connections are different. I found this scheme:

[image]

Maybe you @spacehuhn can help me to look at the right spot. I dont know where to look at.

cnu77, 2020-02-04:
Thanks @alican
> I have the same board and I tried the following steps: https://github.com/puckk/CJMCU-3212
> More information is here: [spacehuhn/wifi_ducky#91](https://github.com/spacehuhn/wifi_ducky/issues/91)

I tried both the links.

As I am new to these h/w parts, its difficult for me to find even the correct pins. I tried to hold the two circled spots together using a pin header, again it fails with the same error.

I guess this has something to do with the basic config like Drivers. But Windows 10 do not even tell if there is any error.

When I connect, I can see the WiFi part can be connected from my mobile. So, I guess the WiFi part is working, but I am not sure if the same WiFi part is deducted by my Laptop or not.

Also, I am not sure if the GPIO0 and GND are already connected (hardwired) or not as there is no mapping diagrams. There was a link but seems to be broken.

Fingers crossed.

cnu77, 2020-02-06:
Now, I tried this video https://www.youtube.com/watch?v=G-Z2rf_AN-c

I followed the steps.

1. Flash ATMEGA32U4, using Step1.ino file
2. Flash ESP8266 using esp8266_flasher.exe and esp8266_wifi_duck_4mb.bin as bin file. Held a jumper cable to the 2 circles, pushed it to USB. Connected to the USB. Did not take the cable out. Click on Download on the flasher and uploaded and finally said 'leaving...'

Now, I can connect to the WiFi Duck with password quackquack.

But, I cannot run the script. Nothing happens.

TheMMcOfficial, 2020-03-01:
I have the same issue the old wifi_ducky works but the wifiduck doesn't. With the wifi_ducky I can run ducky script only in I gues the US layout. For the SD card on the CJMCU-3212 I don't know if you can use it and why I didn't test it.

TheMMcOfficial, 2020-03-04:
After checking the repo page I just realise taht the layout of the wifi_ducky plugs: 

Arduino | ESP82666
-- | --
TX | RX
RX | TX
GND | GND
VCC (3.3V) | VCC (3.3V)

and the wifi_duck use:

ESP8266 | Atmega32u4
-- | --
D1 alias GPIO 5 | 3 alias SCL
D2 alias GPIO 4 | 2 alias SDA
GND | GND

[image]

If we refer to this picture and look at the wiring we use the wifi_ducky and not the wifi_duck so we need to adapt the code if it's possible....

armaan11, 2020-05-09:
Did anyone figure this out yet?

alican, 2020-05-10:
I'm also waiting for progress. I'm sad that I can't use the board. I bought it just to play with USB ducky

armaan11, 2020-05-20:
> I'm also waiting for progress. I'm sad that I can't use the board. I bought it just to play with it with wifiduck.

You can use the board using the older wifiducky software at the moment.

kariez, 2020-05-24:
Please share if you've found the necessary modifications to the code for this particular board.

TheMMcOfficial, 2020-05-25:
You have to use this repo because of the wiring of the PCB. 
https://github.com/spacehuhn/wifi_ducky

kariez, 2020-05-25:
Yes, i know the old versio works fine. I even managed to change the layout but i would really like this updated firmware. To change the layout, you just need to modify the code in keyboard.cpp according to your own layout. File is located in C:\Program Files (x86)\Arduino\libraries\Keyboard\src

As far as i know, the SD card can only be used to store scripts. It's pretty pointless since they can already be stored in SPLIFFS. Using it as a mass storage would be awesome, but i highly doubt it's possible due the contruction of the board.

TheMMcOfficial, 2020-05-25:
If you look at the code you will find that it's impossible to have the new functions on this board it use something like master slave between the ESP8266 and the Arduino (ATMEGA32U). The CJMCU-3212 is not plug like this like I mentioned in my last posts... The ESP have more space than the arduino so you can reprogram the keyboard function and add some layout but the arduino will not have enough space ... If you succeeded I need to take a look at it because I just add the mouse and sd card option and my memory is almost full.

michalmonday, 2020-06-15:
WifiDuck is using I2C protocol, that is available on specific pins. If you take a look at the schematic of the CJMCU-3212 board, you can notice that these pins of Atmega32U4 are available as header pins. On Esp12-F these pins are not connected to anything but they're big enough to easily solder them. So if you have a soldering iron, you could solder these 2 connections yourself.  

[image]

alican, 2020-06-15:
> 
> 
> WifiDuck is using I2C protocol, that is available on specific pins. If you take a look at the schematic of the CJMCU-3212 board, you can notice that these pins of Atmega32U4 are available as header pins. On Esp12-F these pins are not connected to anything but they're big enough to easily solder them. So if you have a soldering iron, you could solder these 2 connections yourself.
> 
> [image]
> 
> Edit:
> It looks like there's a way to choose between both: I2C and serial communiation (using Serial1 which seems to be connected properly on the CJMCU-3212).
> 
> In config.h you can uncomment `// #define ENABLE_SERIAL`
> https://github.com/SpacehuhnTech/WiFiDuck/blob/master/atmega_duck/config.h

@michalmonday 
Thank you Michal, you are the hero we are waiting for.

michalmonday, 2020-06-15:
Not really a hero:p. It appears that someone already made an [issue about using serial](https://github.com/SpacehuhnTech/WiFiDuck/issues/41).  

Btw I just tested it (I don't have CJMCU-3212 but I have similar board that connects Atmega32U4 and Esp8266 using "Serial1" port of Atmega32U4). That's what I did:
- uncommented  `// #define ENABLE_SERIAL` in [esp_duck/config.h](https://github.com/SpacehuhnTech/WiFiDuck/blob/master/esp_duck/config.h)  
- selected settings like that (using 2.7.1 Esp8266 for Arduino library):  
[image]  
- clicked `Sketch->Export compiled binary` (which produced "bin" file)  
- uploaded "bin" file to my board using [nodemcu-flasher](https://github.com/nodemcu/nodemcu-flasher) (idk how to do it for CJMCU-3212 but I guess there are tutorials online)  
- uncommented all the lines below from [atmega_duck/config.h](https://github.com/SpacehuhnTech/WiFiDuck/blob/master/atmega_duck/config.h)  
```
// #define ENABLE_DEBUG
// #define DEBUG_PORT Serial
// #define DEBUG_BAUD 115200

// #define ENABLE_SERIAL
```
- uploaded it to the Atmega32U4 board (it runs on 5V, I selected Arduino Leonardo board setting)  
- replugged the board, connected to the access point, went to `192.168.4.1` address in browser on my mobile phone  
- I opened serial monitor (selected 115200 setting)  
- I ran "STRING test" script and noticed that serial monitor indicates that messages are received by Atmega32U4 properly (but for some reason keystrokes are not typed)  
- after some testing I realized that adding `#include <Keyboard.h>` at the top of atmega_ducky.ino (and reuploading the code) solves that problem (I have no idea why, and I have no idea if it breaks some other things, but that's how I got it working)

alican, 2020-06-16:
It worked! Thanks! :)

>     * after some testing I realized that adding `#include <Keyboard.h>` at the top of atmega_ducky.ino (and reuploading the code) solves that problem (I have no idea why, and I have no idea if it breaks some other things, but that's how I got it working)

I had the same problem and your solution worked for me. But I changed just the position because `#include "keyboard.h"` was already their and I put it on top of `atmega_ducky.ino`

cipz, 2020-06-19:
> 
> It worked! Thanks! :)
> 

Were you able to test this solution on the CJMCU-3212?

alican, 2020-06-19:
> 
> > 
> > It worked! Thanks! :)
> > 
> 
> Were you able to test this solution on the CJMCU-3212?

Yes. it works on the CJMCU-3212 and I run successfully a script.

TheMMcOfficial, 2020-06-23:
does the multiple layout are usable ? You talk about adding the `#include <Keyboard.h>` ?
@alican

TheMMcOfficial, 2020-09-23:
I follow your steps the keyboard work but not the layout with the command LOCAL.... So the big purpose to get this version over the wifi ducky is gone...

TheMMcOfficial, 2020-09-29:
I forked the original repo (wifi_ducky) and added some functions.The website have a similar look to this project (WIFIDuck) and the sdcard can be use to store duckyscript. I store them with the extenxion .txt.

https://github.com/TheMMcOfficial/CJMCU-3212-wifi_ducky

TheMMcOfficial, 2020-09-30:
Sorry the LOCALE with the layout name in CAPITAL (LOCALE FR) work I will check if I'm able to add the function to execute script from the SD Card!

TheMMcOfficial, 2020-09-30:
I'm not sure if it will be possible because after I added the SD function the space left for the variables in the memory are over 100%...

Tapity04, 2022-01-27:
> Not really a hero:p. It appears that someone already made an [issue about using serial](https://github.com/SpacehuhnTech/WiFiDuck/issues/41).
> 
> Btw I just tested it (I don't have CJMCU-3212 but I have similar board that connects Atmega32U4 and Esp8266 using "Serial1" port of Atmega32U4). That's what I did:
> 
> * uncommented  `// #define ENABLE_SERIAL` in [esp_duck/config.h](https://github.com/SpacehuhnTech/WiFiDuck/blob/master/esp_duck/config.h)
> * selected settings like that (using 2.7.1 Esp8266 for Arduino library):
>   [image]
> * clicked `Sketch->Export compiled binary` (which produced "bin" file)
> * uploaded "bin" file to Esp8266 using [nodemcu-flasher](https://github.com/nodemcu/nodemcu-flasher) (idk how to do it for CJMCU-3212 but I guess there are tutorials online)
> * uncommented all the lines below from [atmega_duck/config.h](https://github.com/SpacehuhnTech/WiFiDuck/blob/master/atmega_duck/config.h)
> 
> ```
> // #define ENABLE_DEBUG
> // #define DEBUG_PORT Serial
> // #define DEBUG_BAUD 115200
> 
> // #define ENABLE_SERIAL
> ```
> 
> * uploaded it to the Atmega32U4 (it runs on 5V, I selected Arduino Leonardo board setting)
> * replugged the board, connected to the access point, went to `192.168.4.1` address in browser on my mobile phone
> * I opened serial monitor (selected 115200 setting)
> * I ran "STRING test" script and noticed that serial monitor indicates that messages are received by Atmega32U4 properly (but for some reason keystrokes are not typed)
> * after some testing I realized that adding `#include <Keyboard.h>` at the top of atmega_ducky.ino (and reuploading the code) solves that problem (I have no idea why, and I have no idea if it breaks some other things, but that's how I got it working)

Hello! This helped me perfectly with the CJMCU-3212.

As @michalmonday says I have just edited "#define SERIAL_BAUD 9600" AND "Serial" in the "Protocol" tab instead of "I2C"
Flashed ESP with nodemcuflasher.
I lost a few hours due to the lack and spread of information, for this device the jumpers cables are to bootload the ESP module, then you unplug it and flash the ATMEGA (the ones that are next to the USB at the edge of the PCB (This will cause the ESP module to not start the network once connected) so be sure to unsolder/disconnect that connection after you do the flashing.

This was my experience running WifiDuck with the CJMCU-3212
Hope this helps anyone struggling with this hardware

Z3R0NULL, 2022-06-17:
> Hello! This helped me perfectly with the CJMCU-3212.
>
> As @michalmonday says I have just edited "#define SERIAL_BAUD 9600" AND "Serial" in the "Protocol" tab instead of "I2C" Flashed ESP with [...]

Hello you get a yellow mensaje that "Internal Conecction Problem" in yellow color idk why.