SpacehuhnTech/WiFiDuck #31: [Question] Can I use the .bin upload of the Wifi Ducky to flash the new Duck bin

craftycram, 2020-01-24:
Hey,
is it possible to use the .bin upload feature of the original WifiDucky to upload the new WifiDuck bin to the ESP without having to unsolder everything?
Cheers,
craftycram

spacehuhn, 2020-01-24:
You can in theory yes. You'd have to select serial instead of i2c in the board/compiler settings and export a .bin file yourself and also flash the atmega with the new firmware.
But I would only recommend doing this if you know how to reflash the esp *by hand* in case something fails.

craftycram, 2020-01-24:
okay, yeah i already ordered some new chips, but it will take some time for them to arrive in Germany from China..
If it goes wrong I have one of these ESP-12F flashing-boards (I think you had one in one of your videos), should be able to flash it with that as well, but I managed to get a pretty good looking end result with the wires and chips together which I don't really wanna destroy ^^ 
In which file are the I2C/serial settings located at?
I'm relatively new to flashing only the chip, not a nodemcu.

closed 2020-05-31 by stale[bot]