SpacehuhnTech/WiFiDuck #35: [Question] Generic ESP32 and Arduino Pro Micro - Internal Connection Problem (PT Brasil)

feelixh, 2020-02-09:
Link com vídeo do problema: https://photos.app.goo.gl/LaXWtYSDqBmR7ikp6

Estou usando a seguinte conexão: https://photos.app.goo.gl/Uw55MrXU4AacEBag6

Placa selecionada para o ESP8266: https://photos.app.goo.gl/Uw55MrXU4AacEBag6

Placa selecionada para o atmega: https://photos.app.goo.gl/99vVTME4REXJDH197

Foto adicional: https://photos.app.goo.gl/Nt5YKK93Em7dJTC38

A seguinte mensagem aparece quando tento executar um script pelo webserver: "**Internal Connection Problem**"

Alguém sabe no que estou errando?

Bommlon, 2020-06-03:
Does the problem still occur using the current version of the code? If yes, please explain your problem in English.
It would greatly help us figuring out a solution.

Nikakz, 2020-07-16:
Congratulations on the project, as I am noob I would like to know how to add the Portuguese keyboard PT-BR, I did not understand the steps well. Thanks

Bommlon, 2020-07-16:
Please open a separate issue because this problem doesn't seem to have anything in common with this issue.

closed 2020-11-21 by stale[bot]