SpacehuhnTech/WiFiDuck #37: Russian layout

YuriMKarelin, 2020-02-23:
Hello. I'd like to add a russian layout to your amazing project. I have wrote file locale_ru.h and made all needed changes as you explain. But instead russian word, for example, "мама" (mother in english) I see word "fVfFfVfF". Letters V and F on russian layout is in place of russian letters м and а, but why they are in english and what means a small letter f before each of them?
I have attached my file, but Githab have wanted me to change extention to .txt. Initially it was .h.

[locale_ru.txt](https://github.com/spacehuhn/WiFiDuck/files/4242411/locale_ru.txt)

spacehuhn, 2020-06-03:
Hi, sorry for such a late reply. I didn't have enough info to act initially and then forgot. But I now took a few minutes and added your keyboard layout. I can reproduce the behaviour. I can only suspect some kind of trouble with UTF-8 characters. I'll look into it further later.

spacehuhn, 2020-09-08:
Finally came around to fixing it (https://github.com/SpacehuhnTech/WiFiDuck/commit/eb41fbcab6f25dc37ab1ec9a3890730a49cc29d7)! 🎉

closed 2020-12-09 by spacehuhn