SpacehuhnTech/WiFiDuck #38: wifiducky send and receive

zhjygit, 2020-03-15:
I study it about one years and have tried many version or applicaiton.
what i matter most is that, whether the wifiducky can send and receive data between target and host pc. 
Can you solve it ????Attention, with no network else, just esp8266.
Nearly, most of wifiducky of github just could send commands from host to target, but it cannot download datas or files from target pc, just lile nc to nc ,ftp to ftp, c&c, even meterpreter of kali.
Now, what i do is as follows, under c/s model, plant a client.exe on the target, and communicate with it via server.exe on attact host. I finish it.
With no network else, do you have other way to communication (send and receive)???

zhjygit, 2020-03-15:
“You could make the target download executables from the Wi-Fi chip, instead of the internet. Or execute different attacks and send the results back to the Chip. Or open up a reverse shell on the ESP8266s Wi-Fi”
Just like the problems you said ago @spacehuhn

Bommlon, 2020-06-03:
You could create a grabify link and use the terminal to open it from the target. With a bit of ingenuity you could even exfiltrate small amounts of data by adding arguments to the URL.

closed 2020-09-08 by stale[bot]