SpacehuhnTech/WiFiDuck #45: CLI debug

ruspea, 2020-06-13:
Hello, Bit of a shot in the dark

When creating a file in serial with debug enabled using the commands (create example.duck) then (write example.txt "STRING Hello World!") it creates the file perfectly and reports success!

When I then run that file form serial ie (run "example.txt") the first HID output is correct i get "Hello world" but the following run requests from serial output over HID as "**STRING** Hello World" until the esp is reset and it will run correctly once more until another reset!

I have come in to this issue after adding mqtt to cli and using mqtt to create/remove/run scripts - it works great the first time round the following runs always include the "STRING" prefix until the esp is reset again

I just wanted to know if this is something that's been designed as a feature as I am using it in debug mode and if so is there a way i can work around it.

if I create the file in the webui I can run it from serial just fine as many times as i want its just when running a file **created in the serial console** that only works without including the word STRING in the output once.

I did double check and found this is also true in the current build of wifi ducky! Im happy to share my mqtt version with you if any of this sounds interesting!

it uses wifi manager connects to a mqtt server and waits for a payload however can only be used once before a reboot :D

Bommlon, 2020-06-15:
Maybe you are missing a linebreak after "hello world"?
The write command doesn't add an appending linebreak by itself.

spacehuhn, 2020-06-15:
I just tested it myself and 

```
create test
write -f test -c "STRING something"
write -f test -c "STRING else"
cat test
```

prints
`STRING somethingSTRING else`

That means `write` is not appending linebreaks by it's own.
What you can do is either to include a linebreak yourself or use the `stream <file>`, `close` and `read` commands.
With those you can open a file stream and everything you send over serial will be added as is to the file, until you send the `close` command.

ruspea, 2020-06-15:
Thanks for getting back to me. 
I did wish to try this but I had a problem closing the stream, sending the word close was also written to the scream from serial however I can use the close command in the web cli. 

Is there a way round this. Thanks

spacehuhn, 2020-06-15:
Probably the same issue, you need to send `close` with and after a linebreak like:

```
stream somefile
.....your text.......
close
```

but if you send it without linebreaks, the device sees it as this:

```
stream somefile.....your text.......close
```

closed 2020-09-20 by stale[bot]