SpacehuhnTech/WiFiDuck #46: help error running script!

shooter47, 2020-06-13:
I am running wifiduck on a Debkit NodeMcu v3 and I have an error, the error is from the attached image.
[image]
[image]
[image]

jLynx, 2020-06-13:
https://github.com/SpacehuhnTech/WiFiDuck/blob/master/README.md#hardware

shooter47, 2020-06-13:
> https://github.com/SpacehuhnTech/WiFiDuck/blob/master/README.md#hardware

friend I already did it like this manual, I just want help!

jLynx, 2020-06-13:
You didn't post any photos on how you wired it up

spacehuhn, 2020-06-15:
The error tells you that the communication between both microcontrollers (ESP8266 and Atmega32u4) has issues.
Maybe check the wiring and all connections?

We need more information in order to help you out @shooter47 
What are the steps you made to create the device?
Did you follow the steps closely https://github.com/SpacehuhnTech/WiFiDuck/blob/master/README.md#hardware?

shooter47, 2020-06-23:
> El error le dice que la comunicación entre ambos microcontroladores (ESP8266 y Atmega32u4) tiene problemas.
> ¿Quizás verifique el cableado y todas las conexiones?
> 
> Necesitamos más información para ayudarte @ shooter47
> ¿Cuáles son los pasos que hiciste para crear el dispositivo?
> ¿Seguiste los pasos de cerca https://github.com/SpacehuhnTech/WiFiDuck/blob/master/README.md#hardware ?

thank you very much friend, you are great @spacehuhn. I already found that the problem was in the wiring, I just have a question of how can I use it with an esp8266 01, regards.

spacehuhn, 2020-06-23:
You can edit the pins used for i2c in [`espduck/config.h`](https://github.com/SpacehuhnTech/WiFiDuck/blob/master/esp_duck/config.h#L23) or you [enable serial](https://github.com/SpacehuhnTech/WiFiDuck/blob/master/esp_duck/config.h#L23) and use RX and TX pins.

closed 2020-06-24 by shooter47