SpacehuhnTech/WiFiDuck #47: Connect to existing WiFi

justusiv, 2020-07-03:
In my use case i would prefer the WiFi duck connect to my home WiFi instead of creating a WiFi that devices connect to. I have setup the ESP8266 to connect to my WiFi to host a web page, but would not have the skill set to get the WiFi duck to do it. 
Thoughts.

aroslanov, 2020-09-05:
Same question. I bought WIFI Duck to control my KVM switch using local WIFI. Is there a workaround exists or we'll have to rewrite part of the code?

esjaycc, 2020-09-25:
Would love this feature.

I also want to controll my KVM switch.

unarat, 2020-10-20:
I made some changes that work. Instead of creating an AP, it will now connect to an existing AP using the SSID and PWD in the config.h file. It will not be difficult to add some settings that will allow a creation or connection of an AP.

Changes at lines 82 & 181
[webserver.cpp.txt](https://github.com/SpacehuhnTech/WiFiDuck/files/5407997/webserver.cpp.txt)

Script.js change
Line 117:   ws = new WebSocket("ws://" + location.hostname + "/ws");

After changing the Script.js you will have to run webconverter.py

Jorgen-VikingGod, 2021-02-23:
I managed to do it this way (in caseyour network could not resolve hostnames)

Connecting WiFi and save the current IP into a file called `config.js`

```cpp
  WiFi.begin(STA_SSID, STA_PASSWORD);
  if (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.printf("STA: Failed!\n");
    WiFi.disconnect(false);
    delay(1000);
    WiFi.begin(STA_SSID, STA_PASSWORD);
  }
  saveConfigToJSON();
```

```cpp
void saveConfigToJSON() {
  // Delete existing file, otherwise the configuration is appended to the file
  SPIFFS.remove("/config.js");
  // Open file for writing
  File file = SPIFFS.open("/config.js", FILE_WRITE);
  if (!file) {
    Serial.println(F("Failed to create file"));
    return;
  }
  file.print("var ip = '" + WiFi.localIP().toString() + "';");
  debugln("var ip = '" + WiFi.localIP().toString() + "';");
  // Close the file
  file.close();
}
```

in all html files using script.js, like `index.html` and `terminal.html` please  load `config.js` before loading all other js files

```html
  <script src="config.js"></script>
  <script src="script.js"></script>
  <script src="index.js"></script>
```

On top of script.js add these lines:

```js
var address = ip;
``` 

in method `ws_init` use this address

```js
function ws_init() {
  status('connecting...');
  ws = new WebSocket('ws://' + address + '/ws');
```

This way the ESP modifies the `config.js` file each time it becomes a new address and the website can connect the websocket to the right server.

RayPS, 2021-07-03:
+1 I want this feature.

Thanks to @Jorgen-VikingGod I've able to get it working.
But the more simpler solution I did on the second step is just this line
```JavaScript
ws = new WebSocket("ws://" + location.host + "/ws");
```
So no need to saveConfigToJS and other modification to the html.

Jorgen-VikingGod, 2021-07-03:
@RayPS awesome - so simple is the world  :+1: 
I'll adapt my code as well. I didn't read the doc of location.host - thought it was only the hostname; but it handles the ip address too. 
What an easy and smart solution :partying_face: 

Thx for sharing your findings!

justusiv, 2022-01-25:
I have yet to try this but since i just got my duck working thanks to the other threads i will be giving this a go. I will say that i love the way wled https://github.com/Aircoookie/WLED handles this scenario. if it finds the wifi you setup it will connect to that if not it will spin up an AP. sort of the best of both worlds and would not force to to provide connection details at the time of flashing. I am not smart enough to figure out how that have done it but maybe someone here would be be able to.

justusiv, 2022-01-25:
@RayPS @Jorgen-VikingGod  any chance of a fork i am having a hard time knowing exactly what goes where.

EmileSpecialProducts, 2025-04-14:
I have made a [fork](https://github.com/EmileSpecialProducts/UltraWiFiDuck) that will have an AP and in the setting page you can specify the WiFi connection.
So that you can connect it to you WiFi network.