SpacehuhnTech/WiFiDuck #50: When I connect to the wifi duck it says that there is an internal connection problem

M-a-c-a-r-o-o-n, 2020-08-08:
**Describe the error**
After connecting to the duck there is a yellow bar at the top saying that there is an internal connection problem. There is no output to the serial monitor and I can't type anything using the duck

**Environment:**
 - OS: Windows 10
 - Version: 1.0.4 (0)
 - Hardware: Arduino Pro Micro, Weimos D1 Mini, Spacehuhn's PCB

**Attempts**
I have tried connecting to the serial monitor without success and I have checked all the solder connections which are fine

Does anyone have any idea what the issue might be?

M-a-c-a-r-o-o-n, 2020-08-08:
I've tried connecting them with jumper wires and I still get the same issue.

Nikakz, 2020-08-12:
> Tentei conectá-los com fios de jumper e continuo com o mesmo problema.

Hello, I have the same problem, found a solution?

spacehuhn, 2020-09-07:
Hi. It would be really helpful if you could share pictures of your setup, so we can see the hardware connections.

Nikakz, 2020-09-07:
> Oi. Seria muito útil se você pudesse compartilhar fotos de sua configuração, para que possamos ver as conexões de hardware.
Hello, thank you for your answer, I was able to solve using the wemos D1 mini, with esp8266 and node mcu, the connection is not stable, I think for energy reasons. Now I'm having doubts in the deauther, because I couldn't get the led and the oled to work, I think I'm not sure what to comment on in the aconfig.h code, the coexion is working well. Thanks

hunny1389, 2020-09-21:
[image] 
[image]
[image]

Hi, i also got the same issue as @M-a-c-a-r-o-o-n .. Please find the connection diagram attached. Also as @Nikakz said, my WS2812B is not working even when i had selected LED as ON. The diagram for the LED connection is not attached . Also in case the image is not visible

Arduino               Wemos D1 Mini
    2                              D2
    3                              D1
   RAW                          5v
   GND                        GND    

Also, i had to install EasyAsyncTCP and EasyAsyncWebServer from github as the same were not available on arduino library, which were of the latest version. The library version might be the issue ?

hunny1389, 2020-09-22:
Also, I was able to get it working once but once I get the internal connection error, the device remains in the same state. And after reinstalling wifi duck it works again.

spacehuhn, 2020-09-23:
The connections seem correct. But it could be an issue with the jumper cables or the breadboard. Maybe one of them has a loose contact or something like that.

Hydrostevo, 2020-09-23:
Hallo,
I have the same issue, the wiring is good and I can get a serial message on the ATMega...
> Interpreting:  Replying with status
> RECEIVED [SOT]  '             [EOT]
but that gets into a loop and then disconnects.  I am using cheap boards so that is probably causing it.  Going to try with a Wemos D1.

Tschüss

hunny1389, 2020-10-13:
@spacehuhn I had thought the same. But it happens everytime. I had permanently soldered all the connection and check it with a multimeter. In some cases I got it working if I disconnected the two wires except the power connections and reconnected them back. But not always possible. Will try it again when I get some new wemos board. Btw has someone used this with other boards like nodemcu ? If possible lemme know

hunny1389, 2020-10-14:
So an update to the above... the code is working fine for NODEMCU 12E version. SO you guys can switch. Btw i'll be using the board for the whole day tommorow and give you all an update

Nikakz, 2020-10-14:
Hello, when I had this problem, but it was with a esp12 and then with an MCU node, it was not with the wemos D1 mini, it was a matter of energy that the esp consumes and is generating this page as you reported and I gave up on the esp 12 and node mcu and already with wemos I had stability and there were no problems with the connection.

Nikakz, 2020-10-14:
> **Descreva o erro**
> Após conectar ao pato, há uma barra amarela na parte superior informando que há um problema de conexão interna. Não há saída para o monitor serial e não consigo digitar nada usando o pato
> 
> **Meio Ambiente:**
> 
> * SO: Windows 10
> * Versão: 1.0.4 (0)
> * Hardware: Arduino Pro Micro, Weimos D1 Mini, PCB do Spacehuhn
> 
> **Tentativas**
> tentei conectar ao monitor serial sem sucesso e tenho verificado todas as conexões de solda que são bons
> 
> Alguém tem ideia de qual pode ser o problema?
Hello, when I had this problem, but it was with a esp12 and then with an MCU node, it was not with the wemos D1 mini, it was a matter of energy that the esp consumes and is generating this page as you reported and I gave up on the esp 12 and node mcu and already with wemos I had stability and there were no problems with the connection.

hunny1389, 2020-10-19:
Sadly... i am facing issues with the nodeMCU too :'(.. It worked fine for some hours n it stopped working.

Nikakz, 2020-10-19:
> Infelizmente ... estou enfrentando problemas com o nodeMCU também: '(.. Funcionou bem por algumas horas n parou de funcionar.
Hello, as I spoke with esp12 and node mcu for me it didn't work, I only got stability with the wemos D1 mini and the neopixel led works, but it only stays with the green kuz on, it doesn't light up blue and I couldn't solve it, the mine is stable on the connection.
[image]
[image]
[image]
[image]

KatzeMau, 2020-11-15:
@spacehuhn Same issue like @hunny1389 on the NODE MCU for Az-Delivery 
(Pictures)
[image]
[image]
[image]

spacehuhn, 2020-11-16:
@KatzeMau You missing the Atmega32u4: [README.me#hardware](https://github.com/SpacehuhnTech/WiFiDuck#hardware)

KatzeMau, 2020-11-16:
@spacehuhn can you please provide a link for an esp8266 with a atmega32u4 chip ?

Bommlon, 2020-11-23:
Right now there isn't a cheaper option that is as compatible as the DSTIKE WIFI Duck

KatzeMau, 2020-12-01:
> Right now there isn't a cheaper option that is as compatible as the DSTIKE WIFI Duck

Yeah, bought one :)

closed 2021-03-08 by stale[bot]

slash128v6, 2022-01-22:
Have there been any updates on this? I am having the same "Internal Connection Problem" issue. I have tried with a Pro Micro and D1 Mini as well as a genuine Leonardo and ESP8266 NodeMCU.

spacehuhn, 2022-01-27:
Have a look here: https://github.com/SpacehuhnTech/WiFiDuck/issues/94#issuecomment-1021038551

30oSo18, 2023-02-22:
hola tengo el mismo problema tengo este aparato q compre no se si me pueden dar una mano 
[image]
[image]

PranayDhoble11, 2023-03-12:
hello, I have the same problem

luxkatana, 2024-01-01:
Hey 👋 I am a littlebit confused. So if I have an esp8266 then I don't need the Atmega32u4?

luxkatana, 2024-01-02:
> Não
> 
> Sandra Viana
> Enviado do meu iPhone
> 
> 
> 
> > Hey 👋 I am a littlebit confused. So if I have an esp8266 then I don't
> > need the Atmega32u4?

So I do need the Atmega32u4?