SpacehuhnTech/WiFiDuck #51: locale ES: Impossible to write apostrophe (')

andreswhr, 2020-08-17:
Testing the ES layout on my WiFiDuck:

```
REM es char test
LOCALE GB
GUI r
notepad.exe
ENTER
DELAY 200
STRING !"#$%&'()*+,-./0123456789: =>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_abcdefghijklmnopqrstuvwxyz{|}~¿¡ñÑçÇºª€·¨`
ENTER

```
returns the following in a notepad:

!"#$%&.()*+,-./0123456789: =>?@ABCDEFGHIJKLMNOPQRSTUVWXYZ[\]^_abcdefghijklmnopqrstuvwxyz{|}~¿¡ñÑçÇºª€·¨`

As you can see, every character is translated correctly except for the 7th one, where the apostrophe (') is translated into a full stop (.)

This is making it impossible for me to type powershell commands into the wifiduck, as that character is commonly required.

Any advice will be appreciated. Thanks in advance.

Bommlon, 2020-08-17:
You can try to fix this by modifying the keyboard layout, [here's how](https://spacehuhn.wiki/wifiduck/development/translatekeyboardlayout/).

spacehuhn, 2020-09-07:
Thank you for the report. Seems like our translator made a little mistake. Luckily I was able to find the bug (https://github.com/SpacehuhnTech/WiFiDuck/commit/ca3a1975c31f342da7db8e5bd554fc30dbb30f67).

closed 2020-09-07 by spacehuhn