SpacehuhnTech/WiFiDuck #62: new .bin official releases

MrCook1es, 2021-02-01:
Hi big boss,
since precompiled .bin v1.0.0 release, you've made several updates to the sources.
https://github.com/SpacehuhnTech/WiFiDuck/releases
i read that 43 changes have been made, would you like to re-run the sources and publish a .bin v1.0.1?
so we can upgrade easily directly from the web interface ♥

RithyNicolasTAN, 2021-04-16:
Hi,
Thanks for your application.
Like @MrCook1es, I've bought a WiFiDuck on Aliexpress and I want to upgrade it with the lastest version including FR locale.
Can you publish a new bin in order to upgrade easily from the web interface ?
Thanks

MrCook1es, 2021-07-15:
no response from developer?

spacehuhn, 2021-10-20:
Hi, I'm prepping a new release. But just as a heads-up: you'll need to flash both the atmega (through Arduino IDE) to get the new keyboard layouts and the ESP8266 (through the web interface).

closed 2022-04-27 by stale[bot]