SpacehuhnTech/WiFiDuck #63: Internal Connection Problem Yellow warning when trying to run any script on the WiFi Duck from DSTike

Matrod, 2021-02-06:
Just received a pre-built version of the WiFiDuck from Travis Lin. When connected, it seems to work and allows association to the wifiduck hotstop, nonetheless when trying to execute any script, pressing the run button or enabling the auto-run feature, a yellow warning appears saying "Internal Connection Problem". As I said, this is a pre-built model and not the separate boards featured on other topics with the same error message. I tried to import an existing ducky script as well, which works perfectly on the Rubber Ducky, and it does not work on the WiFiDuck either.

I also tried re-flashing both parts of the firmware, but I was able to do just one of them. The other (atmega) does not respond, according to the instructions on the videos. It is not even recognized in the Arduino IDE interface as a valid port, which is strange, since it works and as I said, allows association to the default WiFiDuck hotspot.

When issuing the command "status" on the terminal window of the tool, it generates the same error. Any help would be appreciated. Tried on Windows, Mac and Linux without results. 

Is there a way to run any diagnostics to rule out a hardware failure?

Thanks for your help.

winselstute, 2021-02-27:
Hi all.

I have exactly the same problem. My self-soldered wifi duck(y) worked well with the old wifi ducky software. So I think, my the hardware is not the problem (Beetle USB ATMEGA32U4 Mini + 3,3V regulator + ESP 12 E, without separate LED/Neopixel).

Now I flashed the new wifi duck files like described in the actual description. 

I can connect to the wifi duck, can edit and save some sample scripts but when trying to run, the yellow error message "Internal Connection Problem" appears. When reconnecting, it says "connected" again.

Any idea?

Thanks for your help.

winselstute, 2021-03-01:
Sorry, my fault. Old Wifi Ducky used RX/TX and new Wifi Duck uses SDA/SCL. Resoldered the connectiona and voila, it works like a charm.

h3cker10632, 2021-04-07:
i have the same problem and i dont know what to do i also have a self soldered  WIFI-ducky but i checked the connections over 100 times and they are connected correctly.  I have also reuploaded the software 2 times to the boards, i have a pro-micro with the needed mega32u4 board and a nodemcu esp8266 boards.

Any idea on how i should fix this??
Thank you

aniketpy, 2021-05-16:
i have same issue  i am using arduino pro micro and esp8266 12E board when I try to edit script it shows internal connection error in yellow.
Please help..
Thank you

closed 2021-08-21 by stale[bot]