SpacehuhnTech/WiFiDuck #65: My Malduino W Displays: Internal Connection Error. Please Help Me This Is Driving Me Crazy.

S1m1B1lly, 2021-03-17:
Does Someone Know What To Do When Malduino W Doesnt Stop Saying Internal Connection Error?
Pls  Help

S1m1B1lly, 2021-03-22:
fixd

closed 2021-03-22 by S1m1B1lly

huntman312, 2021-03-31:
how did you fix this?

huntman312, 2021-03-31:
> 
> 
> fixd

how did you fix this?

S1m1B1lly, 2021-03-31:
Just email Support@maltronics.com with any issue.
for me they sent two files.
it didnt work becuase a mismatch between the  versions of the chips and the software.
Email them and they tell you how.

jay-nutt3r, 2021-08-25:
@huntman312 don't suppose you have them files still?

jay-nutt3r, 2021-08-25:
@S1m1B1lly Don't suppose you still have the files?

jay-nutt3r, 2021-08-25:
@huntman312 did you also fix this issue?

huntman312, 2021-08-25:
I believe i still do. But i wont be home for a while to check (like 6 hours). But if you email Support@maltronics.com he will get back to you with the files and instructions probably sometime really early in the morning if you live United States.

MonsieurBark, 2022-08-28:
same pb Internal Connection Error and the support don't respond :(

boxxyhf, 2022-09-20:
> I believe i still do. But i wont be home for a while to check (like 6 hours). But if you email [Support@maltronics.com](mailto:Support@maltronics.com) he will get back to you with the files and instructions probably sometime really early in the morning if you live United States.

Could you perhaps still share the two files?

lukariosmen, 2022-09-20:
[malw-06-07-2022 (1).zip](https://github.com/SpacehuhnTech/WiFiDuck/files/9610741/malw-06-07-2022.1.zip)
Hi. Here are the needed files :).

boxxyhf, 2022-09-20:
> [malw-06-07-2022 (1).zip](https://github.com/SpacehuhnTech/WiFiDuck/files/9610741/malw-06-07-2022.1.zip) Hi. Here are the needed files :).

Thank! That fixed it. Working as a charm again!

Fox8ter, 2022-10-17:
Hello @lukariosmen , (or @boxxyhf )

How to use these files?
can you provide the instructions ?

REgards

boxxyhf, 2022-10-18:
Hello, @Fox8ter,

Follow this guide: http://docs.maltronics.com/malduino-w/troubleshooting/
But use the files in @lukariosmen's post.

Fox8ter, 2022-10-18:
Same for the uf2 file?

Fox8ter, 2022-10-18:
Yeah same... All works now... Really thx.

MeKAniml, 2023-03-31:
how do you use the uf2 as the the exe just let you choose .bin files ?
btw , i got the error right after following the reset guide ^^

BlackCube4, 2023-11-19:
I have the same problem :/