SpacehuhnTech/WiFiDuck #67: New WifiDuck compatibility with WHID Cactus?

IMExperts, 2021-04-09:
Is the new WifiDuck firmware compatible with WHID Cactus? Is there any tutorial on how to flash it in WHID? All I can find is flashing the old Wifi_ducky tutorials on the web. I had to buy WHID just because of the unavailability of Wifi Duck in my region.

IMExperts, 2021-04-09:
@spacehuhn please look into my question. thanks

Bommlon, 2021-04-18:
This was answered in [issue #54](https://github.com/SpacehuhnTech/WiFiDuck/issues/54#issuecomment-731774999). Look into it and see if it helps.

redxhawk, 2021-04-23:
> This was answered in [issue #54](https://github.com/SpacehuhnTech/WiFiDuck/issues/54#issuecomment-731774999). Look into it and see if it helps.

Tried it but it didn't work.

closed 2021-07-30 by stale[bot]