SpacehuhnTech/WiFiDuck #71: `serial.serialutil.SerialTimeoutException: Write timeout` when flashing ESP-12E

didymus, 2021-08-06:
I just followed the directions given. ATMEGA flashed fine. COM is there.

**Expected behavior**
Honestly, after ATMEGA flashed perfectly fine I expect this to work. 

```
Executable segment sizes:
IROM   : 386708          - code in flash         (default or ICACHE_FLASH_ATTR) 
IRAM   : 28052   / 32768 - code in IRAM          (ICACHE_RAM_ATTR, ISRs...) 
DATA   : 1292  )         - initialized variables (global, static) in RAM/HEAP 
RODATA : 4764  ) / 81920 - constants             (global, static) in RAM/HEAP 
BSS    : 26672 )         - zeroed variables      (global, static) in RAM/HEAP 
Sketch uses 420816 bytes (40%) of program storage space. Maximum is 1044464 bytes.
Global variables use 32728 bytes (39%) of dynamic memory, leaving 49192 bytes for local variables. Maximum is 81920 bytes.
esptool.py v2.8
Serial port COM11
Connecting...
Traceback (most recent call last):
  File "C:\Users\didymus\Documents\ArduinoData\packages\wifiduck\hardware\esp8266\2.6.1/tools/upload.py", line 69, in <module>
    esptool.main(cmdline)
  File "C:/Users/didymus/Documents/ArduinoData/packages/wifiduck/hardware/esp8266/2.6.1/tools/esptool\esptool.py", line 2890, in main
    esp.connect(args.before)
  File "C:/Users/didymus/Documents/ArduinoData/packages/wifiduck/hardware/esp8266/2.6.1/tools/esptool\esptool.py", line 475, in connect
    last_error = self._connect_attempt(mode=mode, esp32r0_delay=False)
  File "C:/Users/didymus/Documents/ArduinoData/packages/wifiduck/hardware/esp8266/2.6.1/tools/esptool\esptool.py", line 455, in _connect_attempt
    self.sync()
  File "C:/Users/didymus/Documents/ArduinoData/packages/wifiduck/hardware/esp8266/2.6.1/tools/esptool\esptool.py", line 394, in sync
    timeout=SYNC_TIMEOUT)
  File "C:/Users/didymus/Documents/ArduinoData/packages/wifiduck/hardware/esp8266/2.6.1/tools/esptool\esptool.py", line 337, in command
    self.write(pkt)
  File "C:/Users/didymus/Documents/ArduinoData/packages/wifiduck/hardware/esp8266/2.6.1/tools/esptool\esptool.py", line 300, in write
    self._port.write(buf)
  File "C:/Users/didymus/Documents/ArduinoData/packages/wifiduck/hardware/esp8266/2.6.1/tools/pyserial\serial\serialwin32.py", line 323, in write
    raise writeTimeoutError
serial.serialutil.SerialTimeoutException: Write timeout
the selected serial port serial.serialutil.SerialTimeoutException: Write timeout
 does not exist or your board is not connected
```

**Environment (please complete the following information):**
 - OS: Win11 Pro 22000.120
 - Arduino IDE 1.8.15
 - CJMCU/DM-3212

**Attempts**
- Downgraded the ESP-12E Wifi Duck boards version 
- Tried as `Generic ESP8266` instead of `Node MCU ESP-12E` (which it is)
All of these changes still produce same error posted above.

**Additional context**
I went on the 8266 forum and maybe the module itself is bad? I'm not sure. This is the first Arduino thing I've messed with.

Maybe there is supposed to be some timeout specified in the code that ArduinoIDE looks for? I will look into this further after an unrelated appointment I have in a few minutes.

TQMatvey, 2021-09-29:
same lmao

Tapity04, 2022-01-27:
Hi! this is talked about in this discussion #30

closed 2022-08-12 by stale[bot]