SpacehuhnTech/WiFiDuck #74: is there any cli method rather then a web interface to execute the scripts?

siddharthkhandelwal1, 2021-08-16:
> Please search for existing (open and closed) issues to avoid duplicates.

justusiv, 2022-01-25:
When i read this comment i see more of like an access via API. I could see all kinds of fun use cases for it with automation, and it would really allow this to interface with anything and allow people to code against it in their language of choice. A fun random example that pops into my mind is one that unlocks my computer when i get home and locks it when i leave.

jstorrs, 2022-01-28:
Here's a python script that runs /myscript:

    import asyncio
    from websockets import connect
    
    uri = "ws://192.168.4.1/ws"
    
    async def wifiduck_run(script):
        async with connect(uri) as websocket:
            await websocket.send(f"run {script}")
            await websocket.recv()

    asyncio.run(wifiduck_run("/myscript"))

justusiv, 2022-01-28:
Thank you for posting that @jstorrs
I was able to get it to work. Being able to write or update a script on the fly would be the other piece that would be very handy to have in the tool bag. Looks like there is a function "create" and "write" that would be useful.

justusiv, 2022-01-30:
After a few minutes playing around i was able to get this to work. I did run into issues with spaces and new lines until i did this. 
`data = """STRING 123
STRING 456
"""`
and
`        await websocket.send(f'write {script} "{data}"')`

closed 2022-08-12 by stale[bot]