SpacehuhnTech/WiFiDuck #83: internal connection error

TQMatvey, 2021-09-29:
internal connection error

**Environment (please complete the following information):**
Windows 11
 chrome latest
 dstike
a lot

and i reflashed it a lot of ways

21r8390, 2021-10-01:
I have the same error. ⚠
I use a Malduino W. I already tried to reload the firmware on it. Don't get any errors during this process. 
But as soon as I enter #status in the terminal, I get as error message: _internal connection error_ 

The UI and the settings works fine, but creating a file doesent.
WifiDuck Version: 1.0.4 (4)

TQMatvey, 2021-10-02:
Hi, i was able to fix it by switching the wifi duck into esp mode, flashing, then back to normal mode. Everything works as new!!

TQMatvey, 2021-10-02:
Wait how did you got 1.0.4 if the latest is 1.0.0

21r8390, 2021-10-03:
Never mind... I just reinstalled everything and made sure the connectors are plugged in. 

I don't know what the issue was but now everything works as expected. 🤔

TQMatvey, 2021-10-03:
Nice that everything works :) i wish you a good day or night!!

closed 2021-10-03 by TQMatvey