SpacehuhnTech/WiFiDuck #84: SD Card Module support too and installation method!

sameerakhtari, 2021-10-04:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
SO the problem with Wifi duck is that you can pass limited line of script. Adding a SD Card module with it will help us to use it as simple USB Rubber Ducky if needed.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
We can use DIP Switch or something like that in order to change it from WIFI DUCK to RUBBER DUCKY. And as far the program concern there should be a method that check if certain pins (eg. pin 8 and pin 9) are connected by dip then Act as a RUBEER DUCKY . Else act as a WIFI DUCK.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
I had seen another repo of someone who was attaching SD Card module with Pro Micro to make a simple Rubber Ducky. If that can happen and this can happen then why not both can happen in same project :)

**Additional context**
Add any other context or screenshots about the feature request here.
If you don't use this template, your issue might be closed and tagged invalid!

SSBhaumik, 2022-01-31:
This feature would be terrific.

Z3R0NULL, 2022-06-22:
I would like this

sameerakhtari, 2022-07-28:
Hey @spacehuhn I tried to merge DuckDuino (rubber ducky)
(https://github.com/Seytonic/Duckduino-microSD.git) 
with your wifi duck program. but for some reason my memory is getting overflowed. Can you do it in some way that I would able to run both SD Card Module and ESP12F with a single Arduino Pro Micro , and able to switch between them by using Dip Switch.?