SpacehuhnTech/WiFiDuck #85: Together not working

hexan86, 2021-10-20:
**Describe the error**
ATmega is not powering ESP8266, but the parts alone are working.
If I connect the ATmega is recognized as keyboard and the D1 mini has network and web interface working.
But if I connect ONLY the ATmega, the ESP8266 is not powering up.

**To Reproduce**
Followed all the instructions, connected correctly the wiring (tested with the multimeter) and flashed correctly (the parts alone are perfectly working).

**Expected behavior**
See the wifiduck network

**Environment (please complete the following information):**
Hardware: 
- Arduino Pro Micro ATMega32U4
- D1 Mini Pro ESP8266 ESP12

**Attempts**
RAW and 5V are connected, checking with the multimeter on the D1 pins (5V and GND) I can see there is voltage 4,53V, but it isn't working.

In one (and only one) mattempt to see the voltage on the D1, multimeter red on ATmega RAW and black on D1 GND, I saw the voltage and the D1 LED briefly went on.

spacehuhn, 2021-10-21:
It could be that the pro micro has a fake power regulator or other fake parts in the power circuit that create the problem.

hexan86, 2021-10-21:
> It could be that the pro micro has a fake power regulator or other fake parts in the power circuit that create the problem.

So what do you suggest? Try to buy a regulator or some other solution? 

Thanks!

spacehuhn, 2021-10-21:
Yes you could try that. 
If you end up building a new WiFi Duck, use some jumper wires to test you can power both boards correctly first.

hexan86, 2021-10-21:
> Yes you could try that. If you end up building a new WiFi Duck, use some jumper wires to test you can power both boards correctly first.

Yeah, thanks for the advice. I have a little experience in electronics and soldering, the project seemed so simple that I feeled extremely confident and I went straightforward to the assembling, without testing before.

If you want, you can close the issue (is not a fault in your project) and if I retrieve a power regulator or another wifiduck, I'll let you know if there are any other issues.

p3ppi, 2022-06-07:
Describe the error
ATmega (Pro Micro) is not powering ESP8266 (nodemcu). 
If I connect the ATmega is recognized as Leonardo and the nodemcu has network and web interface working (if powered itself).
But if I connect ONLY the ATmega, the ESP8266 is not powering up, though when plugging the ATmega in sometimes the built-in LED of the nodemcu flashes once.
When powering ATmega and nodemcu seperatly I can connect to wifiduck network. But there is an internal connection error tho the ATmega show on the webinterface.

To Reproduce
Followed all the instructions, controlled several times, connected correctly the wiring (tested with the multimeter) and flashed as described on e.g. https://wifiduck.com/ and basically all other sources.

Expected behavior
work as intented in manual

Environment:
Hardware:
    Arduino Pro Micro ATMega32U4
    nodemcu different versions

Software:
  Win11 Arduino IDE 1 (as IDE 2 gives an error on upload)

closed 2022-12-23 by stale[bot]