SpacehuhnTech/WiFiDuck #93: [Locale]:

Litchyi, 2021-11-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### What layout are you translating?

CH

### Where do you need help?

Hi, could you add the Swiss keyboard layout based on the german one because it's the same. Thanks and i love your project (sorry bad english).

spacehuhn, 2021-11-28:
Hi. What are the differences to the German layout?

Litchyi, 2021-11-28:
> Hi. What are the differences to the German layout?
The only difference (if I'm not mistaken) is that the abbreviations (ctrl, alt etc ...) are not in English on the German keyboard. But it's just the writing on the keyboard and it doesn't influence the code.

Litchyi, 2021-12-01:
i think you can just copy and rename the german folders

Litchyi, 2021-12-14:
After verification, there are differences because switzerland has several languages ​​... Sorry for my mistake.

spacehuhn, 2022-01-25:
Just added the CH-FR layout: https://github.com/SpacehuhnTech/WiFiDuck/blob/master/atmega_duck/locale_chfr.h
```
LOCALE ch-fr
DELAY 5000
STRING -,.'§<$0123456789aàbcdeèéfghijklmnopqrstuvwxyz_;:!?"()*/&%°+=>£AäBçCDEFGHIJKLMNOöPQRSTUüVWXYZ[]{}@\#¬|¦~¢€`´^âÂÄÀáÁêÊëËÈÉîÎïÏìÌíÍôÔÖòÒóÓûÛÜùÙúÚ
```
and CH-DE: https://github.com/SpacehuhnTech/WiFiDuck/blob/master/atmega_duck/locale_chde.h
```
LOCALE ch-de
DELAY 5000
STRING -,.'§<$0123456789aäbcdefghijklmnoöpqrstuüvwxyz_;:!?"()*/&%°+=>£AàBçCDEèéFGHIJKLMNOPQRSTUVWXYZ[]{}@\#¬|¦~¢€`´^¨âÂÄÀáÁêÊëËÈÉîÎïÏìÌíÍôÔÖòÒóÓûÛÜùÙúÚ

```

closed 2022-01-29 by Litchyi