SpacehuhnTech/WiFiDuck #94: internal connection problem

Dementora, 2021-11-27:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Describe your problem

i think its because of atmega32u4

### Steps to reproduce

my wiring its ok but it doesn't run any thing and show  internal connection problem error

### What hardware are you using?

pro micro d1 mini

### Anything else?

[image]
[image]

my wiring its ok but it doesn't run any thing and show  internal connection problem error and when i remove config.h line 17 when i run command like:

DELAY 1000
GUI R
STRING CMD
ENTER
in serial monitor  115200 baud it come :

RECEIVED [SOT] 'DELAY 1000
' [EOT]
Interpreting: DELAY 1000
Replying with status {wait: 0,repeat: 0} [04 00 00 00 ]
RECEIVED [SOT] '' [EOT]

i dont know how to fix my wifiduck pls help.

Bl4ckterminator, 2021-11-29:
Hi, same issue need help please.

aklehm, 2021-12-17:
I have the same problem, but I figured out, that it works, if I power both boards via USB and connect the I2C and GND from board to board. 
If I power the ESP board from the AVR board (RAW --> 5V), I also get the "Internal Connection Problem".
The ESP gets a voltage of 4.7V with a current of 80mA and the wifi is working fine, so I would exclude an under-voltage. 
This is far away from a solution, but maybe it helps someone to find one.

prawinrai1993, 2021-12-27:
same issue for me too

prawinrai1993, 2021-12-27:
I see that just one line is received in the buffer. Other lines not receiving.
[image]

ajayamola9, 2021-12-28:
Hi, same issue need help please.

aklehm, 2022-01-10:
Try to increase the first delay (line 19) in the esp_duck.ino file. I tried 10 seconds  (delay(10000);) and it works well for me. This is maybe to long, so you can try to decrease it.
My sugestion is, that the default delay of 200ms is to short before the esp tries to connect to the other controller and in some cases, the controller hasn't finished its I2C setup, so a connection couldn't be established.

I haven't watch the code, but it would be helpful, if there were a reconnect function.

mattybeens, 2022-01-13:
i have this isssue on both of my wifi ducks. one is the dstike and the other is the WeMos D1 Mini and AITRIP Pro Micro ATmega32U4. i tried aklehms method to no avail.

prawinrai1993, 2022-01-14:
changing delay to 1000 works for me on line number 19 esp_duck.ino. Due to initialiasation taking more time

mattBrighto, 2022-01-21:
in `tools >` diable debbugging and change `protocol` to serial. i mean now you dont have internal connection problem but it didnt work either. now you have like `Running /scriptname` but it doesnt run the script

mattBrighto, 2022-01-21:
> in `tools >` diable debbugging and change `protocol` to serial. i mean now you dont have internal connection problem but it didnt work either. now you have like `Running /scriptname` but it doesnt run the script

update: it could not work just on my atmega cuz i kinda broke it, the micro usb port just tore off when i was trying to take the cable out :/

justusiv, 2022-01-22:
> > in `tools >` diable debbugging and change `protocol` to serial. i mean now you dont have internal connection problem but it didnt work either. now you have like `Running /scriptname` but it doesnt run the script

I tried this and appeared to have the same result as you. No "internal connection problem" however my scripts do not type anything as well.

ismail424, 2022-03-28:
> Hi, there seem to be multiple open issues about this so I'm just going to repeat what I found here: #99
> 
> Disabling Debug and setting the protocol to I2C (instead of Auto) on the atmega_duck seems to lead to improvements: [image] [image]

This didn't solve my problem. Is there something else I can do to solve this issue?

uhuhuhuhuhuhuhuh, 2022-05-09:
Yeah is there another method to fix this issue

mducq, 2022-07-22:
Hi,

i have the same problem, i have fixed it with "serial" protocol.
i don't have the message, i can run script without problem but... when i plug my nodemcu (aliexpresse version) on a windows computer, it's seen as serial converter and not as a keyboard.

[image]

do you have an issue for this please?

thanks

sameerakhtari, 2022-07-28:
I am getting same issue, still don't know how can I resolve this but I think this issue is generated by Arduino pro micro(or any other version you are using). cause I programmed it before ESP12F(nodemcu or any other) and Arduino pro micro and it worked fine for a time, then I broke my pro micro port , so I had to change my pro micro , since then, I am getting this error. I haven't programmed ESP12F or wrongly wired them. 
If anyone knows exact way to resolve this error, Kindly Help me out.

dtuando, 2022-08-18:
OK I Figured out the issue. If you bought one recently most likely you will need to reflash the atmega.
Using arduino, add the boards from the GitHub install instructions. 
Flash USING LEONARDO. not the Default DUSTIKE WiFi Duck Board option.

[example](https://void.link/file/4rgBB4Uz0jVQlkQu/pPEcN9r2tqIelOnJ/Untitled.png)

matti377, 2023-01-06:
Is there still no fix?
I keep having the same issue

matti377, 2023-01-06:
ok i found the error. You sould read the docs.
I followed an old guide from Seytonic and these connections arent actual anymore.

PranayDhoble11, 2023-03-12:
Internal connection problem ...........this error is solved
[image]

PranayDhoble11, 2023-03-12:
[image]
[image]
[image]
[image]
[image]

PranayDhoble11, 2023-03-12:
[image]
[image]

TomFang1, 2023-07-07:
Have a look at my fork, maybe my solution also fit for you:
https://github.com/TomFang1/WiFiDuck_CJMCU_3212/blob/master/README.md

Evil0000, 2024-08-03:
my script is not running on computer. it shows running but nothing happens

closed 2025-05-06 by stale[bot]