SpacehuhnTech/WiFiDuck #99: Internal connection  problem

prawinrai1993, 2022-01-14:
https://github.com/SpacehuhnTech/WiFiDuck/blob/40456264ddff330b6c2c4310cfebbb472a0e6458/esp_duck/esp_duck.ino#L19

please update the delay to 1000ms to resolve the internal connection issue

Egor2101, 2022-01-14:
> https://github.com/SpacehuhnTech/WiFiDuck/blob/40456264ddff330b6c2c4310cfebbb472a0e6458/esp_duck/esp_duck.ino#L19
> 
> please update the delay to 1000ms to resolve the internal connection issue
Does not work...

mattybeens, 2022-01-14:
Didnt work for me either :(

prawinrai1993, 2022-01-14:
> Didnt work for me either :(

try increasing delay. I am using nodemcu, for me worked. for other user 10000ms worked, who is using a d1 mini. not facing internal connection issue any more. Also scripts working super smooth.

mattybeens, 2022-01-14:
i tried both of those times. I also have a dstike wifi duck V2 that has the same issue after updating.

justusiv, 2022-01-18:
Tried both the 1000 and 10000 and i am having the same issue.

slash128v6, 2022-01-22:
Same issue for me.

spacehuhn, 2022-01-24:
I did some testing, and with debug enabled, I could reproduce the issue sometimes. Try disabling Debug if it isn't already. 
[image]

slash128v6, 2022-01-24:
I have tried with Debug enabled and disabled. I tried on my DSTIKE board and now that board is no longer working. It appears to be an issue with I2C. I can run scripts via Serial between ESP and 32U4, albeit very slow.

spacehuhn, 2022-01-24:
Using I2C (or serial given the right hardware) instead of the Auto mode also seems to make a difference sometimes. I have a growing suspicion that the atmega32u4 is partly responsible, as it seems to be very sensitive.
[image]

justusiv, 2022-01-25:
ok so that got me working... set both debugs to disable and both protocols to I2C.
Gonna play around but thanks for the info!

slash128v6, 2022-01-25:
That seems to be working for me now too, thank you!

spacehuhn, 2022-01-25:
Hopefully, that's it 🤞 . I updated the Wifi Duck AVR boards so that these options are set by default from now on.

slash128v6, 2022-01-25:
So far so good, I've reflashed a few times just to confirm and it's still working on both my DSTIKE board and DIY setup. Appreciate the help!

slash128v6, 2022-01-25:
I just tried on an ESP8266 connected to genuine Arduino Leonardo and it did not work, unfortunately. But the other setups are still working. Just wanted to let you know.
________________________________
From: Jason Granat
Sent: Monday, January 24, 2022 4:19:14 PM
To: SpacehuhnTech/WiFiDuck; SpacehuhnTech/WiFiDuck
Cc: Comment
Subject: Re: [SpacehuhnTech/WiFiDuck] Internal connection problem (Issue #99)

So far so good, I've reflashed a few times just to confirm and it's still working on both my DSTIKE board and DIY setup. Appreciate the help!

mattybeens, 2022-01-25:
https://www.amazon.com/dp/B08BJTS2N8?ref=ppx_yo2_dt_b_product_details&th=1
https://www.amazon.com/dp/B07S5HL54T?psc=1&ref=ppx_yo2_dt_b_product_details

Mines not working either :(

slash128v6, 2022-01-25:
> https://www.amazon.com/dp/B08BJTS2N8?ref=ppx_yo2_dt_b_product_details&th=1 https://www.amazon.com/dp/B07S5HL54T?psc=1&ref=ppx_yo2_dt_b_product_details
> 
> Mines not working either :(

try disabling debug on both sketches and hard set protocol to I2C on both sketches. That got me working with the same hardware from Amazon.

mattybeens, 2022-01-25:
> try disabling debug on both sketches and hard set protocol to I2C on both sketches. That got me working with the same hardware from Amazon.

I got it, i was dumb and had to use the "generic esp8266"  board

prawinrai1993, 2022-01-25:
The solution works fine with disabling debug and selecting i2c. Also my board required generic esp8266 selection in ide before flash. works for both d1 wemos mini and esp8266.

closed 2022-01-25 by prawinrai1993