SpacehuhnTech/WiFiDuck #100: doesnt run scripts

markrharvey, 2022-01-19:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Describe your problem

whenever i try to run a script it tells me that it is running but nothing happens 

### Steps to reproduce

go to  wifi.duck 
scroll down to script 
click run 
banner changes from connected to running 
nothing happens on computer that it is pluged into 

### What hardware are you using?

DIY Build (D1 mini and pro micro)

Ciph3rCat, 2022-02-09:
Same issue here with cjmcu-3212, any help?

Z3R0NULL, 2022-06-16:
> Same issue here with cjmcu-3212, any help?

Hii i flashed this on my CJMCU-3212 but theres say in Yellow "Internal Connection Problem" why? you have solution ?

closed 2022-12-23 by stale[bot]