SpacehuhnTech/WiFiDuck #101: Only first line runs

slash128v6, 2022-01-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Describe your problem

Only first line of script runs

### Steps to reproduce

1. Create script.
2. Click run.
3. Web interface shows Internal Connection Problem.
4. Serial monitor shows Interpreting: and the first line of the script.
5. If I make the first line of the script LED 255 255 255 and run the RGB LED changes to white, so it appears that line executes, but no further lines execute.

### What hardware are you using?

DYI Build (Pro Mini + D1 Mini)

slash128v6, 2022-01-24:
I just tried flashing this same firmware to a DSTIKE board that I purchased. After flashing the Arduino code the DSTIKE now only executes the first line of any script and status shows "Internal connection problem". So now the DSTIKE that I purchased is non-functional.

prawinrai1993, 2022-01-25:
try referring to #94 to resolve the issue. I had same issue before.

closed 2022-08-12 by stale[bot]