SpacehuhnTech/WiFiDuck #103: Any problems with mac os keyboard profiler?

SSBhaumik, 2022-01-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Your Question

Does the DIY wifi duck run seamlessly on mac os given the settings prescribed in your video tutorial or we need to do something about apple hexcode ?

spacehuhn, 2022-01-31:
In Arduino when flashing atmega_duck, you can select the Keyboard USB VID/PID from a list of predefined devices
[image]

SSBhaumik, 2022-01-31:
Thank you so much you're like an excellent mentor, i have great respect for you. Although I don't know a lot about coding but your tutorials are easy to understand.

SSBhaumik, 2022-01-31:
Can i use this apple id on windows too?

SSBhaumik, 2022-01-31:
Ok tested it works on windows, pro micro and d1 mini setup like yours. After flashing was showing internal connection error but i reset my d1 mini then it was running fine.

closed 2022-01-31 by SSBhaumik