SpacehuhnTech/WiFiDuck #109: DHCP not working on WHID injector

125K, 2022-04-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Describe your problem

I flashed WiFiDuck on a WHID injector, but when I connect to the AP, the ESP8266 won't provide me any IP, so I have to set it manually. It's like the DHCP service won't work properly. Also, the DNS server won't work.

### Steps to reproduce

I made a few changes for making this code work on a WHID injector: https://github.com/125K/WiFiDuck-whid

1. Use the esp8266Programmer.ino to be able to flash esp_duck code
2. Flash the esp_duck code
3. Flash the atmega_duck code
4. Now, you can't connect to the network, unless you set an IP manually.

It's like the DHCP server is not working.

### What hardware are you using?

WHID injector

HugoMskn, 2022-08-06:
Hey ! thanks very much for your work, I can finally run wifi duck on my whid, except, when I tried going onto the web page with the IP 191.168.4.1 I have ERR_ADDRESS_UNREACHABLE error, do you know why and could be get in touch on discord or sum ? 
here's mine just in case : Hugo ;)#0667

adamff1, 2022-08-07:
@HugoMskn I have the same error, I don't know why this happens 🤔. If you find some way to fix it, feel free to make a pull request

closed 2023-03-25 by stale[bot]