SpacehuhnTech/WiFiDuck #124: I CAN RUN SCRIPS BUT THEY STOP - CJMCU 3212 - INFO

Z3R0NULL, 2022-12-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Describe your problem

I CAN RUN SCRIPS BUT THEY STOP MIDWAY

### Steps to reproduce

i got a CJMCU 3212 the only badusb that can be obtained in argentina besides attiny85
I flashed it successfully, sometimes it can run a script, but sometimes it stops in the middle of the script... IDK WHY

### STEPS
I flashed ESP with the bin file, okay. I just uncommented line 16 in esp_duck/config.h like this
```
/*! ===== Communication Settings ===== */
#define ENABLE_SERIAL
```

because if I don't uncomment that line it gives me an error ` INTERNAL CONNECTION PROBLEM`  and it doesn't execute scripts
ok now i flashed the ATMEGA with atmega_duck.ino

I connect to wifiduck's wifi and when I try to run the script it works fine, but sometimes, and sometimes not.

### EXAMPLE SCRIPT RUNNING
```
LOCALE ES
DEFAULTDELAY 500

WINDOWS r
STRING notepad.exe
ENTER
STRING HELLO WORLD!
```
I open the serial monitor in atmega and it gives me this. I notice that when this line appears. the script stops
`Replying with status {wait: 1,repeat: 0} [04 01 00 00 ]` 

as you can see it appears ` {wait 1,` 
but i realised that this is wrong because when a script sucessfully run it says wait 0
` Replying with status {wait: 0,repeat: 0} [04 00 00 00 ]` 

### SUCESSFULLY SCRIPT RUN EXAMPLE DEBUG OUTPUT SERIAL MONITOR ATMEGA - "{WAIT: 0,"
```
12:26:15.604 -> RECEIVED [SOT] '' ...
12:26:15.651 -> RECEIVED '1001100 1001111 1000011 1000001 1001100 1000101 100000 1000101 1010011 1010 ' [EOT]
12:26:15.884 -> Interpreting: LOCALE ES
12:26:16.024 -> Replying with status {wait: 0,repeat: 0} [04 00 00 00 ]
12:26:16.209 -> RECEIVED [SOT] '' ...
12:26:16.255 -> RECEIVED '1000100 1000101 1000110 1000001 1010101 1001100 1010100 1000100 1000101 1001100 1000001 1011001 100000 110101 110000 110000 1010 ' [EOT]
12:26:16.675 -> Interpreting: DEFAULTDELAY 500
12:26:16.861 -> Replying with status {wait: 0,repeat: 0} [04 00 00 00 ]
12:26:17.048 -> RECEIVED [SOT] '' ...
12:26:17.094 -> RECEIVED '1010 ' [EOT]
12:26:17.187 -> Interpreting: 
12:26:17.187 -> Replying with status {wait: 0,repeat: 0} [04 00 00 00 ]
12:26:17.421 -> RECEIVED [SOT] '' ...
12:26:17.469 -> RECEIVED '1010111 1001001 1001110 1000100 1001111 1010111 1010011 100000 1110010 1010 ' [EOT]
12:26:17.700 -> Interpreting: WINDOWS r
12:26:18.304 -> Replying with status {wait: 0,repeat: 0} [04 00 00 00 ]
12:26:18.536 -> RECEIVED [SOT] '' ...
12:26:18.582 -> RECEIVED '1010011 1010100 1010010 1001001 1001110 1000111 100000 1101110 1101111 1110100 1100101 1110000 1100001 1100100 101110 1100101 1111000 1100101 1010 ' [EOT]
12:26:18.999 -> Interpreting: STRING notepad.exe
12:26:19.737 -> Replying with status {wait: 0,repeat: 0} [04 00 00 00 ]
12:26:19.924 -> RECEIVED [SOT] '' ...
12:26:19.970 -> RECEIVED '1000101 1001110 1010100 1000101 1010010 1010 ' [EOT]
12:26:20.155 -> Interpreting: ENTER
12:26:20.713 -> Replying with status {wait: 0,repeat: 0} [04 00 00 00 ]
12:26:20.898 -> RECEIVED [SOT] '' ...
12:26:20.993 -> RECEIVED '1010011 1010100 1010010 1001001 1001110 1000111 100000 1001000 1000101 1001100 1001100 1001111 100000 1010111 1001111 1010010 1001100 1000100 100001 1010 ' [EOT]
12:26:21.461 -> Interpreting: STRING HELLO WORLD!
12:26:22.158 -> Replying with status {wait: 0,repeat: 0} [04 00 00 00 ]
```
as you can see this time it opened the notepad and successfully writed HELLO WORLD! but lets see whats happen sometimes

### FAIL SCRIPT RUN EXAMPLE DEBUG OUTPUT SERIAL MONITOR ATMEGA - "{WAIT: 1,"
```
12:26:15.604 -> RECEIVED [SOT] '' ...
12:26:15.651 -> RECEIVED '1001100 1001111 1000011 1000001 1001100 1000101 100000 1000101 1010011 1010 ' [EOT]
12:26:15.884 -> Interpreting: LOCALE ES
12:26:16.024 -> Replying with status {wait: 0,repeat: 0} [04 00 00 00 ]
12:26:16.209 -> RECEIVED [SOT] '' ...
12:26:16.255 -> RECEIVED '1000100 1000101 1000110 1000001 1010101 1001100 1010100 1000100 1000101 1001100 1000001 1011001 100000 110101 110000 110000 1010 ' [EOT]
12:26:16.675 -> Interpreting: DEFAULTDELAY 500
12:26:16.861 -> Replying with status {wait: 0,repeat: 0} [04 00 00 00 ]
12:26:17.048 -> RECEIVED [SOT] '' ...
12:26:17.094 -> RECEIVED '1010 ' [EOT]
12:26:17.187 -> Interpreting: 
12:26:17.187 -> Replying with status {wait: 0,repeat: 0} [04 00 00 00 ]
12:26:17.421 -> RECEIVED [SOT] '' ...
12:26:17.469 -> RECEIVED '1010111 1001001 1001110 1000100 1001111 1010111 1010011 100000 1110010 1010 ' [EOT]
12:26:17.700 -> Interpreting: WINDOWS r
12:26:18.304 -> Replying with status {wait: 0,repeat: 0} [04 00 00 00 ]
12:26:18.536 -> RECEIVED [SOT] '' ...
12:26:18.582 -> RECEIVED '1010011 1010100 1010010 1001001 1001110 1000111 100000 1101110 1101111 1110100 1100101 1110000 1100001 1100100 101110 1100101 1111000 1100101 1010 ' [EOT]
12:26:18.999 -> Interpreting: STRING notepad.exe
12:26:19.737 -> Replying with status {wait: 1,repeat: 0} [04 01 00 00 ]
```

now as you can see it stopped at mid of the script saying the ` {wait: 1,`  and the page says ` RUNNING /SCRIPT` in blue  idk why i would like more help if you want more debug log i do more test THANKS A LOT!! sorry for my bad english :D

### What hardware are you using?

CJMCU 3212

### Anything else?

i think it maybe cause with the times or something like that idk... i tried to change the esp_duck.ino line 19 delay line its default in 200 i changed to 1000 and it dont worked. but i dont know how the code works so i changed it with no idea what i was doing.

Z3R0NULL, 2022-12-04:
i hope @spacehuhn reply to me i would like to get it working.
i posted the ATMEGA debug serial monitor but i wanna see more detailed debug but i dont know how to do it...
i wanna see the ESP debug how i have to do?

ill let this pics of my board
[image]
[image]

Z3R0NULL, 2022-12-04:
i think maybe the problem is in here?? idk :( would like help im givin all the info that i have at this moment

Lines in ESP_DUCK / COM.CPP file
```
typedef struct status_t {
    unsigned int version : 8;
    unsigned int wait    : 16;
    unsigned int repeat  : 8;
} status_t;

namespace com {
    // =========== PRIVATE ========= //
    buffer_t receive_buf;
    buffer_t data_buf;

    bool start_parser         = false;
    bool ongoing_transmission = false;

    status_t status;

    void update_status() {
        status.wait = (uint16_t)receive_buf.len
                      + (uint16_t)data_buf.len
                      + (uint16_t)duckparser::getDelayTime();
        status.repeat = (uint8_t)(duckparser::getRepeats() > 255 ? 255 : duckparser::getRepeats());
    }

    // ========== PRIVATE SERIAL ========== //
#ifdef ENABLE_SERIAL
    void serial_begin() {
        debugsln("ENABLED SERIAL");
        SERIAL_COM.begin(SERIAL_BAUD);
    }

    void serial_send_status() {
        update_status();
#ifdef ENABLE_DEBUG
        debugs("Replying with status {");
        debugs("wait: ");
        debug(status.wait);
        debugs(",repeat: ");
        debug(status.repeat);
        debugs("} [");

        for (int i = 0; i<sizeof(status_t); ++i) {
            char b = ((uint8_t*)&status)[i];
            if (b < 0x10) debug('0');
            debug(String(b, HEX));
            debug(' ');
        }
        debugsln("]");
#endif // ifdef ENABLE_DEBUG

        SERIAL_COM.write(REQ_SOT);
        SERIAL_COM.write((uint8_t*)&status, sizeof(status_t));
        SERIAL_COM.write(REQ_EOT);
        SERIAL_COM.flush();
    }

    void serial_update() {
        unsigned int len = SERIAL_COM.available();

        if ((len > 0) && (receive_buf.len+len <= BUFFER_SIZE)) {
            SERIAL_COM.readBytes(&receive_buf.data[receive_buf.len], len);
            receive_buf.len += len;
        }
    }

   // ===== PUBLIC ===== //
    void begin() {
        status.version = 0;
        status.wait    = 0;
        status.repeat  = 0;

        i2c_begin();
        serial_begin();
    }

    void update() {
        i2c_update();
        serial_update();

        if (react_on_status) {
            react_on_status = false;

            debug("Com. status ");

            if (status.version != COM_VERSION) {
                debugf("ERROR %u\n", status.version);
                connection = false;
                if (callback_error) callback_error();
            } else if (status.wait > 0) {
                debugf("PROCESSING %u\n", status.wait);
            } else if (status.repeat > 0) {
                debugf("REPEAT %u\n", status.repeat);
                if (callback_repeat) callback_repeat();
            } else if ((status.wait == 0) && (status.repeat == 0)) {
                debugln("DONE");
                if (callback_done) callback_done();
            } else {
                debugln("idk");
            }
        }
    }
```

Z3R0NULL, 2022-12-06:
i think the error is related by the times or something like that because, it run scrips sometime so it work fine, but sometimes not..
maybe is related to comunication times beetwen esp and atmega i dont know im talking from ignorance.

Z3R0NULL, 2022-12-06:
DAY: 3 - I am trying to communicate with other people who know about the topic. I hope they answer I've been trying to solve it for days and they are my last option.

Z3R0NULL, 2022-12-08:
DAY: 4 - I have learned a lot about the arduino language which is based on C / C++ but I still don't understand the code of the Wifi_Duck project.

Z3R0NULL, 2022-12-09:
DAY: 6 -  I can't, I already tried everything, I still don't want to give up but I don't know what to do anymore. One good thing is that I learned many things.

todely, 2022-12-12:
Hope find solution, i will get the same hardware

Z3R0NULL, 2022-12-13:
DAY: 10 - I think we're going to need help... I will upload a photo of diagram and my hardware modified. 

### IN THIS IMAGE WE CAN SEE PIN DIAGRAM TO FLASH ESP OR ATMEGA
[image]

### HERE MY CJMCU WITH SWITCH
[image]
[image]

i solder a switch in CJMCU to switch between ESP or ATMEGA flash modes. GPIO0 & GND
It is a good hardware because it have 3 modules SDCARD,ESP,ATMEGA but there is not enough information.

Z3R0NULL, 2022-12-13:
> Hope find solution, i will get the same hardware

You can do it work but with the old wifi duck repositori, i dont like the old because no languaje layout like EEUU,FR,AR,ES,JP, etc...

todely, 2022-12-13:
[image]

[image]

something wrong here perhaps ? not same connexion than model, so perhaps problem ?

michalmonday, 2022-12-13:
Hi, could you post your `Communication Settings` section settings from config.h files of both ESP and Atmega? No idea if that's the case but maybe you enabled both: I2C and Serial, and they sometimes interfere with each other because they seem to use the same buffer?

Z3R0NULL, 2022-12-13:
> Hi, could you post your `Communication Settings` section settings from config.h files of both ESP and Atmega? No idea if that's the case but maybe you enabled both: I2C and Serial, and they sometimes interfere with each other because they seem to use the same buffer?

Yeah i think that i enabled Both i2c and serial idk maybe you can help <3

Z3R0NULL, 2022-12-13:
### My Communication Settings

[image]

michalmonday, 2022-12-13:
It looks like only serial is enabled, so that isn't a problem I think.

Z3R0NULL, 2022-12-13:
> It looks like only serial is enabled, so that isn't a problem I think.

the configuration when flashing can give problems?
I'm using this anyway...

### Step1.ino TO FLASH ESP FIRST FLASH THIS IN ATMEGA..
[image]

### ESP FLASH CONF
[image]

### ATMEGA FLASH CONF
[image]

Z3R0NULL, 2022-12-13:
I followed this Tutorial in Chinnese Page 
https://www.wuifan.cn/post/burning-wifiduck-into-cjmcu-3212/

Z3R0NULL, 2022-12-13:
### Nodemcu Flasher to flash esp-wifi-duck.bin CONF
[image]

Z3R0NULL, 2022-12-13:
### I Tried this Changing 115200 to 9600 in Step1.ino But Dont work idk if this is correct...
[image]

todely, 2022-12-13:
For me model gpio (describe on github project for DIY wifiduck) are not the same than on this hardware

todely, 2022-12-13:
For me 

> I followed this Tutorial in Chinnese Page
> https://www.wuifan.cn/post/burning-wifiduck-into-cjmcu-3212/

This is for old duck

Z3R0NULL, 2022-12-13:
> For me model gpio (describe on github project for DIY wifiduck) are not the same than on this hardware
> 
> @Z3R0NULL have you try to change gpio for scl gpio from 5 to 0 in communication settings ?

Yes i already changed im flashing but flashing slow haha i will let you know when it finalizes

Z3R0NULL, 2022-12-13:
> For me model gpio (describe on github project for DIY wifiduck) are not the same than on this hardware
> 
> @Z3R0NULL have you try to change gpio for scl gpio from 5 to 0 in communication settings ?

I just flashed but it's still the same without any changes, it's still working but it stops sometimes... I'll upload a video of my error so you can analyze it better...

Z3R0NULL, 2022-12-13:
> For me model gpio (describe on github project for DIY wifiduck) are not the same than on this hardware

Maybe that's the problem, why the old Wifi-Duck-Old repository works but the new one doesn't?

Maybe you have to adapt the Wifi-Duck-New code to make it work.

We have to look at the Wifi-Duck-Old code and adapt the new one from there... maybe change the GPIO code as you say?

Unfortunately I don't have the knowledge of electronics or arduino programming for that...
@todely 
@michalmonday

Z3R0NULL, 2022-12-13:
> Maybe that's the problem, why the old Wifi-Duck-Old repository works but the new one doesn't?
>
> Maybe you have to adapt the Wifi-Duck-New code to make it work.
>
> We have to look at the Wifi-Duck-Old [...]

I tried to do this a few days ago, but failed.

I tried to pass the code from Wifi-Duck-Old to Wifi-Duck-New...
but it didn't come out

todely, 2022-12-13:
I will check when i will receive hardware but i’m pretty sure probleme come from this.

Z3R0NULL, 2022-12-13:
I just recorded a video.
as you can see in the video the script never ends successfully, but the atmega runs as a keyboard.

I have to say that sometimes the script does finish successfully.

https://youtu.be/wDi8iJEtYrk

I tryed other scripts/payloads but same result.

Here is the script that i want to run in video...

```
LOCALE ES
DEFAULTDELAY 200

GUI r
STRING notepad.exe
ENTER

STRING ------------
ENTER
STRING ---HACKED---
ENTER
STRING ------------
ENTER
STRING ------------
ENTER
STRING ---HACKED---
ENTER
STRING ------------
ENTER
STRING ------------
ENTER
STRING ---HACKED---
ENTER
STRING ------------
CTRL +
REPEAT 40
```

todely, 2022-12-13:
I have look the video ok
Read this
https://github.com/SpacehuhnTech/WiFiDuck/issues/30

It’s our problem

todely, 2022-12-13:
@michalmonday have found same thing than me :i2c is not connected on 3212 but soldering  missing 2 connexion is possible
@michalmonday where is j2 on the board ?

Z3R0NULL, 2022-12-13:
> @michalmonday have found same thing than me :i2c is not connected on 3212 but soldering  missing 2 connexion is possible
> @michalmonday where is j2 on the board ?
> 
> edit: found header pin
> [image]
> 
> need to solder slc and sda to esp to make i2c work
> 
> @Z3R0NULL could you check if these 2 pin are not already connected to esp somewhere ? (With multimeter) 
> 

I'm not at home right now, tell me which pins you want me to check, I have a multimeter...

Draw them for me in the images that I sent you.. i Will check when i get back to home.

todely, 2022-12-13:
We are looking for where slc and sda are on the board, if they are free or connected to something

https://i0.wp.com/randomnerdtutorials.com/wp-content/uploads/2019/05/ESP8266-ESP-12E-chip-pinout-gpio-pin.png?quality=100&strip=all&ssl=1

to make duck work i2c have to been connected between esp and atmega

solution is probably to solder slc and sda from esp to slc and sda on atmega (pin i have mark in red on board picture in my last post, you can try with small « alligator clip » to avoid soldering for nothing)

Z3R0NULL, 2022-12-13:
> We are looking for where slc and sda are on the board, if they are free or connected to something [...]

### I already tested...

I tested SCL PIN from ESP to ATMEGA and it have continuity...

I tested SDA PIN from ESP to ATMEGA and it don't have continuity..

Idk if i tested Well, i Will send photo of the tested pins...

Z3R0NULL, 2022-12-13:
> We are looking for where slc and sda are on the board, if they are free or connected to something [...]

### UPDATE PIN TESTED/CHECK

### SDA PIN

### SCL PIN

todely, 2022-12-14:
Ok so can you connect sda to sda and slc to slc with some grip or anything else ?
I will get it in 10 days i think

todely, 2022-12-14:
I just follow official tutorial for DIY on wifiduck, you see SDA is connected to SDA and SLC to SLC to talk by I2C protocol and not RX/TX
https://github.com/SpacehuhnTech/WiFiDuck#diy-hardware

if you connect wrong pin or touch 5v there is possible damage.
But wait i try by myself, easy to get in my country.

I have to find a smaller iron than the one i have or order small grip by the way.

todely, 2022-12-18:
After this defeat not sure i will tell you if it works !!
😇

todely, 2022-12-19:
Just kidding about football match argentina / french (i’m french)
Not arrive for the moment

Z3R0NULL, 2022-12-19:
> Just kidding about football match argentina / french (i’m french)
> Not arrive for the moment

Oooh i understand now 🤣🤣
here in Argentina people turn crazy in the streets.

todely, 2022-12-19:
not expected before 28/12

Z3R0NULL, 2022-12-25:
Espero tengan una linda navidad.

todely, 2022-12-25:
Happy xmas too !

Z3R0NULL, 2023-01-03:
Yup, so how are you.
Is there any news about that?

todely, 2023-01-03:
Hello !
Item seems to be stuck somewhere in china since 16 december...
[image]

Z3R0NULL, 2023-01-17:
Hello how are you, I decided to take a risk xd.
since I have it saved and if I don't risk it I won't be able to use it.
But I need a little of your help. I only have to solder those wires or what? Can you guide me? thank you very much :D

todely, 2023-01-17:
Hello !
I have check and item is moving from china today (after 1 month stay somewhere...! )
Expecting a delivery in few days (it was a 10 day delivery) 

[image]

If you wanna try you need to test with some electronic grip (better than soldering for testing purpose)
[image]

todely, 2023-01-23:
[image]

Receive, use i2c for communication between esp and atmega
Wifi server is ok
[image]

Does you have a script who doesn't work for you ? (The one in your video have error on "ctrl+", i have to remove this line)

Z3R0NULL, 2023-01-31:
> [image]
> 
> Receive, use i2c for communication between esp and atmega Wifi server is ok [image]
> 
> Does you have a script who doesn't work for you ? (The one in your video have error on "ctrl+", i have to remove this line)
> 
> Edit: Parameter in esp sketch [image]

Hii... did you manage to make it work? does it work fine and no errors?

todely, 2023-01-31:
Everything seems to work but your script (in your video) have an error on line "ctrl+"

Does you have a script not working   For you and you want i try ?

SitrakaResearchAndPOC, 2023-03-06:
For mine, using latest script, even the script doesn't run. Using this tutorial, https://www.youtube.com/watch?v=G-Z2rf_AN-c&pp=ugMICgJmchABGAE%3D, it runs correctly but only english language.

Z3R0NULL, 2023-05-22:
> Hey @Z3R0NULL pudiste darle solucion al problema?

Nope, and you? 
i think the jumper is the only option like todely says in te comments.

TomFang1, 2023-07-07:
Created a fork with install instructions for everyone struggeling with the CJMCU3212:
https://github.com/TomFang1/WiFiDuck_CJMCU_3212/tree/master

SitrakaResearchAndPOC, 2023-07-08:
[TomFang1](https://github.com/TomFang1), please a binary precompiled esp8266 if possible !

PogusTheWhisper, 2024-03-10:
when I try string 1234567890 it just type 1234 and stop, but when I move my mouse it continue to type and stop.

PogusTheWhisper, 2024-03-10:
@Z3R0NULL Any idea what going on?

PogusTheWhisper, 2024-03-10:
I think should make mouse move little bit to fix it, but it doesn't quite make sense.

NULL-BITS, 2024-03-25:
> Any update?

About what?

Z3R0NULL, 2024-06-11:
> Any update?

Hello, I already have my CJMCU 3212 working correctly, you just have to make a jumper as Todely says, from the pins marked in the image SDA TO SDA AND SDL TO SDL from the esp to the atmega

TakuSan22, 2024-09-01:
**hola pudo restableser de fabrica cjmcu3212 me podrian ayudar , o atrabes hard reset .y instalar firmware**
[image]

closed 2025-05-06 by stale[bot]