SpacehuhnTech/WiFiDuck #126: Probleme while flashing Pro Micro

BGM00000, 2023-02-09:
Please I need help when I Upload in the Pro Micro Leonardo it gives me an error, why ?
```
Property 'upload.tool.serial' is undefined
```

[image]

root-x-krypt0n-x, 2023-11-05:
same problem here

root-x-krypt0n-x, 2023-11-06:
> Just flash with normal drivers, not the modified ones. should still work

can you explain what do u mean normal drivers, i just using the original file that i download from this github, but have same problem

SoLongSidekick, 2024-05-12:
> Just flash with normal drivers, not the modified ones. should still work

What does that mean? I didn't download any special drivers for the AVR install and the instructions don't tell you to?

LittleBitSalty, 2024-07-13:
I made a [pull request](https://github.com/SpacehuhnTech/arduino-avr/pull/2) to fix this bug.

> What does that mean? I didn't download any special drivers for the AVR install and the instructions don't tell you 

The "normal drivers" ywaf mentioned are the regular boards included in the Arduino AVR Boards. Although they don't give the same configuration options, they still will work.

closed 2025-05-06 by stale[bot]