SpacehuhnTech/WiFiDuck #134: [Feature Request]: Support ESP32-S2

spuder, 2024-03-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem

It would be nice to support more microcontrollers such as the new line of ESP32-S2/ ESP32-S3 boards as they support keyboard emulation. 

### Solution

The wemos S2 Mini supports keyboard emulation

https://www.wemos.cc/en/latest/s2/s2_mini.html

Others have gotten it working in esphome

https://github.com/esphome/esphome/pull/3917

eadmaster, 2024-03-22:
have you checked this?
https://github.com/wasdwasd0105/SuperWiFiDuck

spuder, 2024-03-22:
Thats exactly what I'm looking for. Just got it working on a Wemos S2. Thank you

closed 2024-03-22 by spuder