SpacehuhnTech/WiFiDuck #136: Internet connection problem

Aryanstha, 2024-05-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Describe your problem

In web "internal connection problem" pop up, scripts is not running

### Steps to reproduce

N/A

### What hardware are you using?

ESP8266

### Anything else?

Maybe issue in wiring connection

f1x1t1x1f, 2024-10-19:
Same issue. The issue apears, when i have a script with GUI or WINDOWS Button. Then the USB restarts and don't run the script.

EmileSpecialProducts, 2025-04-14:
Just use the ESP32-S3  / S2 as this has native USB. 
I have made a [fork](https://github.com/EmileSpecialProducts/UltraWiFiDuck) that also has a Web Programming so you will not even need a IDE to build the software.
And it also supports Bluetooth , Mouse, GPIO Trigger and ESP32-S3,ESP32-S2,ESP32-C3, It will not support the ESP8266