SpacehuhnTech/WiFiDuck #139: Original PCB

Skulo2011, 2024-08-14:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Your Question

I would like to get all original PCB files.

### Description

I want to make the original PCB myself.

LittleBitSalty, 2024-09-02:
If you are talking about the PCB's shown on [this page](https://wifiduck.com/docs/diy/pcb), the download links are provided.  However, if you are talking about the Maltronics [MalDuino W](https://maltronics.com/products/malduino-w), or the [DSTIKE Wifi Duck](https://dstike.com/collections/frontpage/products/dstike-wifi-duck), the PCBs are proprietary.  I have reached out to DSTIKE asking for the designs, as I was interested in making one myself, but, unfortunately I have not gotten a response.  I would recommend using the CJMCU Beetle (or a clone) and the Lolin D1 Mini, as these two combined have a very small profile.  Similar to [this image](https://wifiduck.com/assets/images/ducky_soldered-698307c4da65addf092d498b9682dd4f.jpg).

EmileSpecialProducts, 2025-04-14:
The good thing about this project you do not need custom PCB.
Just use the development bords ( I recommend esp32-s3 devkitc 1) , no SD Card needed for the scripts as it uses the internal flash to store the files.
I have made a [fork](https://github.com/EmileSpecialProducts/UltraWiFiDuck) that also has a Web Programming so you will not even need a IDE to build the software.
And it also supports Bluetooth , Mouse, GPIO Trigger and ESP32-S3,ESP32-S2,ESP32-C3